MarlinFirmware/Marlin #9243: [Question] Build marlin with new gcc versions

Murasame74, 2018-01-19:
Hi all, I'm trying to build Marlin 1.1.8 with avr-gcc 7.2, latest avr-libc and ArduinoIDE 1.8.5 for Anet board.

During compilation process gcc throws this error:

non-constant condition for static assertion

on this define:

#define digitalPinToPCICR(p) ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )

invoked by:

static_assert(digitalPinToPCICR(X_MIN_PIN) != NULL, "X_MIN_PIN is not interrupt-capable");
static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");
static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable");

in file endstop_interrupts.h.

Question is, it's safe to comment-out these instructions ?
If I do it the build process complete fine and I have a nice increment in free space on the avr that let me enable some features I left behind with stock gcc version.

Almost forgot, with stock gcc that comes with ArduinoIDE it builds fine.

Roxy-3D, 2018-01-19:
If it builds OK with the stock gcc...   You can comment out the static_assert's.   They are just there to detect problems with the configuration.

Murasame74, 2018-01-20:
Thanks a lot in the next days I'll do some more tests.

Also a suggestion, isn't better to create a support forum in main marlinfw.org instead of using github issue tracker ? It could help keeping track of "real" issues instead of having all mixed up with user  support questions.

Just my 2 cent.

thinkyhead, 2018-02-14:
> Also a suggestion, isn't better to create a support forum in main marlinfw.org

It is a static HTML site.

closed 2019-07-06 by shitcreek