MarlinFirmware/Marlin #9249: G29 parameter limits

jamhot1, 2018-01-19:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```
Hi all, I am getting a problem running G29 X5 Y5 L25 R65 F110 B160. The procedure does not finish and seems to try and rehome the Z outside of the build plate half way through, then throws a comms error. If I run G29 X4 Y4 L25 R65 F110 B160 ( 4x4 matrix instead it works fine????) This is running latest firmware.

Roxy-3D, 2018-01-19:
It would be easier to help you if we knew which bed leveling system you are using.

jamhot1, 2018-01-19:
Sorry, Yes. in this case it is linear. I tried Bilinear too but it ignores the number of grid points and reverts to what is specified in the default. Using Bilinear also seems to probe a grid that does not match the area specified in the L,R,F,B parameters. It is similar but seems to offset by 5 to 10mm.

Roxy-3D, 2018-01-19:
The Linear bed leveling systems were the original ones...   Most people think the mesh based systems do a better job.    My recommendation is you consider moving to one of those.

jamhot1, 2018-01-19:
I was under the impression that the Mesh system was designed to be a manual procedure? I am using a touch probe to create the points in this case. Am I wrong?

Roxy-3D, 2018-01-19:
The original Mesh system (that proved the concept) was manual.   But that has been expanded.    

If you have a Z-Probe, you might want to try the directions here to bring up UBL:   https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

jamhot1, 2018-01-19:
oh I see! Thanks very much for the guidance. I shall have a look through. I was assuming this information from the code comments. Thank you, I shall have a look through

jamhot1, 2018-01-19:
Ok, I have set to UBL. As soon as i un comment and set to UBL I get the error

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: C:\Users\Dan\AppData\Local\Temp\arduino_build_957883/Marlin.ino.elf section `.text' will not fit in region `text'

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: region `text' overflowed by 20320 bytes

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Anet V1.0.

SwiftingSpeed, 2018-01-20:
That means there isn't enough flash memory to fit the firmware. From what I can tell (don't have a anet board myself) the anet board is a meltzi clone with a smaller atmega processor so I don't think you can fit UBL at all. You could try removing SD card support and maybe even display support, to see if there's even a chance.

jamhot1, 2018-01-20:
ok, Thanks. I am literally going around in circles with this. Can anyone recommend an updated board I could run with this? Something I can run UBL on? I could swear this board has a mind of its own

Roxy-3D, 2018-01-20:
For $20 you can get a complete setup identical to what UBL was developed on:

https://www.ebay.com/itm/3D-Printer-Kit-ATmega-2560-RAMPS-1-4-Board-5pcs-A4988-for-Arduino-RepRap/322934029199?hash=item4b305e3b8f:g:qKIAAOSw71xZnPJD

You probably can just move your LCD display to the RAMPS board.

jamhot1, 2018-01-20:
I was looking at this https://www.aliexpress.com/item/Creality-Motherboard-Controller-Board-Mainboard-for-Creality-CR-10-10Mini-CR-10S-S4-S5-3D-Printer/32849102298.html

Its uses all my existing connectors and I believe Marlin is already used by the creality?

Roxy-3D, 2018-01-20:
Yeah...   That board has an Atmega-2560 on it.    You will have plenty of space for any options you want to turn on.

jamhot1, 2018-01-20:
Great! Thanks for the advice. And just to give some advice back to the community. I've been going around in circles with this partly because my board wasn't taking the firmware properly. I was going from one rev firmware to another testing ideas and ended introducing new issues despite zero changes. I ended up doing a eprom erase and then loading firmware and it cleared an issue. Only took me nearly all day to try.... Talk about frustration. I will however pull the trigger on this new board. Thanks again

boelle, 2019-02-20:
@jamhot1 so you have solved your issue?

maybe close this one?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead