MarlinFirmware/Marlin #9250: FWRetract slow speeds.

Squid116, 2018-01-19:
### Bug Report
- Description: retraction speeds are slow with high values of retraction speed and length.

I recently changed my titan to run as Bowden, and have hit this problem when trying to tube retraction.

- Expected behaviour: retractions occur at speed in retraction settings.

- Actual behaviour: 
Retraction is slow, when higher values of speed and length are used.

- Steps to reproduce:
  - Slice a file using fw retract
  - increase retraction speed and distance and the problem occurs.

Eg 
40mm/s and 10mm causes the issue.
20mm/s and 10mm doesn't.

Using retraction from slic3r these speeds and lengths work fine.

Squid116, 2018-01-20:
I blew away my repo and got latest bugfix 1.1.x and it seems to be going better.  Will continue to monitor and report back.

jameswood, 2018-01-23:
Are you using TMC2130 drivers? I was having the same problem and solved it with tweaks to the current.

Squid116, 2018-02-04:
@jameswood nope, drv8825

I can confirm this is happening... Feels like an overflow issue - (years of Dev experience) but I could be wrong.

Drv8825 32 microsteps.
0.9 degree stepped
3:1 titan extruder
1674 esteps/mm

All of these are pretty high from what I have seen.

Only happens when velocity, length and acceleration are sufficiently high.  I can see a quick extruder move. (jerk?) followed by a long slooooow series of steps till (can be 20+ seconds) the end of the retraction.

Is there some debugging I can insert to help diagnose?

Squid116, 2018-02-04:
@roxy-3d sorry to tag you just to bump this up, but this issue is killing me - could I get some advice on how to get the debugging output you guys would need?

Squid116, 2018-02-05:
Switched to 1/16 microsteps, and can definitely retract more, will play with speed tonight.

Squid116, 2018-06-05:
I think this was a limit of the stepping rate on the mega - I have since upgraded to a re-arm so I am unable to reproduce/test so I'm going to close this off.

closed 2018-06-05 by Squid116

Arismane, 2018-07-16:
@jameswood Can you please tell me what tweaks to the current did you do to solve the problem, because I have the exact same problem with TMC2130 drivers installed and I choose to retract maximum on 25mm/s in order to get rid of it. Thanx in advanced.

jameswood, 2018-07-16:
@Arismane From memory, I think I had to turn the current up. This made the driver eventually overheat, so I added a fan to cool it. Every machine is different, but if it helps at all, I currently have the current set at:
`#define E0_CURRENT         800`

I usually retract 2mm at 40mm/s (Bowden, Titan, E3D v6, 2.85mm)