MarlinFirmware/Marlin #9251: Compiling UBL on RADDS

smoki3, 2018-01-19:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

--------------------
### Compile Error
When I compile with `Autobedlevel UBL` I get an error:
```
In file included from sketch/src/Marlin.cpp:41:0:
sketch/src/module/configuration_store.h:107:43: error: 'E2END' was not declared in this scope
         const static int16_t meshes_end = E2END - 128; // 128 is a placeholder for the size of the MAT; the MAT will always
                                           ^
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).

```
I trying to compile UBL on RADDS. Get these error

Can anyone help?

Roxy-3D, 2018-01-19:
UBL requires the EEPROM size so it knows where (and how many) Mesh's can be stored.

Apparently this isn't known yet on the RADDS 32-bit platforms.    I'll put a link to this message in the main 32-bit thread and see if somebody can rectify that issue...

p3p, 2018-01-19:
I was under the impression that E2END was an Arduino Environment variable so that it would be available on the DUE platform, if that it not the case you need to put an appropriate value in the HAL_DUE.h, from EepromEmulation_Due.cpp it appears that the value should be 4096.

I still need to update the HAL persistent store api to have a size method/constant that will remove the E2END variable from within marlin base code.

AnHardt, 2018-01-19:
RADDS is a shield for the DUE. RADDS has EEPROM on board.

smoki3, 2018-02-20:
https://github.com/MarlinFirmware/Marlin/pull/9737

can we merge this?

Roxy-3D, 2018-02-20:
I merged it...   But I'm wondering if it makes sense to have such a large persistent storage area declared?   8KB is twice as big as what the AVR's have.

closed 2018-06-11 by smoki3