MarlinFirmware/Marlin #9252: RADDS CoreXY no Z Movement

smoki3, 2018-01-19:
------------------------------------------------------------
### Bug Report
- Description: I compiled Marlin 2.x for my RADDS CoreXY. It has a fix mounted probe connected to the z min pin. I Home G28 X0 Y0 works great. If i want to home "G28 Z0" the stepper motor get enabled but there is no movement. 
[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/1647999/Archiv.zip)

Bob-the-Kuhn, 2018-01-19:
It could be because the Z_MIN endstop is always active/triggered.  M119 will tell you if it's enabled and triggered.

smoki3, 2018-01-19:
No endtops are working fine

Bob-the-Kuhn, 2018-01-20:
If you move the Z cable and driver to either the A or B sockets, does it move?

If you have access to some LEDS & wire, try to see if the STEP, ENABLE & DIR signals are working correctly.  Start with a known good socket and then move the LEDs to the Z socket once you've seen life in all three.

sh4jonny, 2018-02-01:
I think i have a same problem. I have minimal speed accel and jerk (3,5,0.4) values, but Z axis not moving. I roll back to old mk4duo firmware and Z axis is ok! I build latest bugfix-2.0.x branch but Z not work ((

sh4jonny, 2018-02-01:
anyone?

sh4jonny, 2018-02-03:
@smoki3 do you fix your Z zxis ?

smoki3, 2018-02-03:
No. I also back to mk4duo. Here everything is fine

sh4jonny, 2018-02-03:
@smoki3 what drivers you use?

smoki3, 2018-02-04:
Tmc2100

thinkyhead, 2018-02-05:
Anyone try troubleshooting, collecting more data? It might help!

smoki3, 2018-02-05:
I already tried to grab some logs. But there is nothing. The display shows z movement forbidden.

sh4jonny, 2018-02-05:
@smoki3 you can try to find some simple drivers a4988 it work for me time after time... drv8825 work almost year and not work now ((( also i try to run on my Eaxis driver - HR4988. But E work, Z no! ((((
Also i take my logic analyzer and HR4988. And can't find somethin special ((( correct DIR STEP and ENABLE signals!!! BUT! in this case my Z ahis is moving!!!! If i unplug usb cable from analyzer it not work.
so... Zaxis now work on a4988
on drv8825 mk4duo is working, marlin don't
Eaxis work ok! in any case
TMC2130 on XY also work in any case.

sh4jonny, 2018-02-05:
@thinkyhead may be we lost some pull-up for Z pins ?

thinkyhead, 2018-02-05:
> The display shows z movement forbidden.

New data!

If you're using `Z_SAFE_HOMING` then you must home X and Y before you can home Z.

smoki3, 2018-02-05:
M119 z is always open

sh4jonny, 2018-02-06:
@thinkyhead i confirm i never use Z_SAFE_HOMING
also my Z endstop work as must. i check it more times.
also i try to remap Z axiz to E1 slot. i check pins by M43 and delete any definition except STEP DIR EN
thank for your help!

sh4jonny, 2018-02-06:
@smoki3 when you have this message - "The display shows z movement forbidden" ?

smoki3, 2018-02-06:
I got it one time. Cant reproduce the message at the moment.

smoki3, 2018-02-06:
Are there any debug feature which we can enable?

smoki3, 2018-02-08:
@thinkyhead

sh4jonny, 2018-02-08:
@smoki3 i think all what we have in this case this is M43 command for debug pins definition. you need enable m43 debug feature in your config.

smoki3, 2018-02-15:
`>>> do_homing_move(Z, -300.00, 0.00)
  current_position=(100.00, 100.00, 0.00) : sync_plan_position
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing`

`PIN: 00        RXD                                    Input  = 1
PIN: 01        TXD                                    Input  = 1
PIN: 02        Z_STEP_PIN                             protected
PIN: 03        Z_DIR_PIN                              protected
PIN: 04        SDSS                                   Input  = 1
PIN: 05        SERVO0_PIN                             Input  = 1
PIN: 06        SERVO1_PIN                             Input  = 1
PIN: 07        HEATER_BED_PIN                         protected
PIN: 08        FAN1_PIN                               protected
PIN: 09        FAN_PIN                                protected
PIN: 10        <unused/unknown>                       Input  = 1
PIN: 11        HEATER_2_PIN                           Output = 0
PIN: 12        HEATER_1_PIN                           Output = 0
PIN: 13        HEATER_0_PIN                           protected
PIN: 14        <unused/unknown>                       Input  = 1
PIN: 15        Z_ENABLE_PIN                           protected
PIN: 16        Y_DIR_PIN                              protected
PIN: 17        Y_STEP_PIN                             protected
PIN: 18        <unused/unknown>                       Input  = 1
PIN: 19        <unused/unknown>                       Input  = 1
PIN: 20        <unused/unknown>                       Input  = 1
PIN: 21        <unused/unknown>                       Input  = 1
PIN: 22        Y_ENABLE_PIN                           protected
PIN: 23        X_DIR_PIN                              protected
PIN: 24        X_STEP_PIN                             protected
PIN: 25        X_CS_PIN                               Input  = 1
PIN: 26        X_ENABLE_PIN                           protected
PIN: 27        Y_CS_PIN                               Input  = 1
PIN: 28        X_MIN_PIN                              protected
PIN: 29        Z_CS_PIN                               Input  = 1
PIN: 30        <unused/unknown>                       Input  = 1
PIN: 31        E0_CS_PIN                              Input  = 1
PIN: 32        Z_MIN_PIN                              protected
PIN: 33        E1_CS_PIN                              Input  = 1
PIN: 34        <unused/unknown>                       Input  = 1
PIN: 35        E2_CS_PIN                              Input  = 1
PIN: 36        Y_MAX_PIN                              protected
PIN: 37        <unused/unknown>                       Input  = 1
PIN: 38        <unused/unknown>                       Input  = 1
PIN: 39        SERVO2_PIN                             Input  = 1
PIN: 40        PS_ON_PIN                              protected
.              SERVO3_PIN                             protected
PIN: 41        <unused/unknown>                       Input  = 0
PIN: 42        <unused/unknown>                       Input  = 0
PIN: 43        <unused/unknown>                       Input  = 0
PIN: 44        <unused/unknown>                       Input  = 0
PIN: 45        <unused/unknown>                       Input  = 1
PIN: 46        <unused/unknown>                       Input  = 1
PIN: 47        <unused/unknown>                       Input  = 1
PIN: 48        <unused/unknown>                       Input  = 1
PIN: 49        E2_ENABLE_PIN                          Output = 0
PIN: 50        <unused/unknown>                       Input  = 1
PIN: 51        E2_STEP_PIN                            Output = 0
PIN: 52        <unused/unknown>                       Input  = 1
PIN: 53        E2_DIR_PIN                             Output = 0
.              MAX6675_SS                             Output = 0
PIN: 54 (A 0)  TEMP_0_PIN                             protected
PIN: 55 (A 1)  TEMP_1_PIN                             Input  = 1
PIN: 56 (A 2)  TEMP_2_PIN                             Input  = 1
PIN: 57 (A 3)  TEMP_3_PIN                             Input  = 1
PIN: 58 (A 4)  TEMP_BED_PIN                           protected
PIN: 59 (A 5)  TEMP_4_PIN                             Input  = 1
PIN: 60 (A 6)  E0_DIR_PIN                             protected
PIN: 61 (A 7)  E0_STEP_PIN                            protected
PIN: 62 (A 8)  E0_ENABLE_PIN                          protected
PIN: 63 (A 9)  E1_DIR_PIN                             Output = 0
PIN: 64 (A10)  E1_STEP_PIN                            Output = 0
PIN: 65 (A11)  E1_ENABLE_PIN                          Output = 0
PIN: 66        <unused/unknown>                       Input  = 1
PIN: 67        <unused/unknown>                       Input  = 1
PIN: 68        <unused/unknown>                       Input  = 1
PIN: 69        <unused/unknown>                       Input  = 1
PIN: 70        <unused/unknown>                       Input  = 1
PIN: 71        <unused/unknown>                       Input  = 1
PIN: 72        <unused/unknown>                       Input  = 1
PIN: 73        <unused/unknown>                       Input  = 1
PIN: 74        MISO_PIN                               Input  = 1
PIN: 75        MOSI_PIN                               Input  = 1
PIN: 76        SCK_PIN                                Input  = 1
PIN: 77        <unused/unknown>                       Input  = 1
PIN: 78        <unused/unknown>                       Input  = 1
`

these are the outputs from the m111 s32 and m43

smoki3, 2018-02-15:
so i now replace the drv8825 with an 4988 now its moving fine. looks like something breaks the drv8825 in the marlin 2 firmware

Bob-the-Kuhn, 2018-02-15:
Marlin only supplies STEP, DIRECTION and ENABLE to the drivers.  

The only possible firmware involvement might be the MINIMUM_STEPPER_PULSE setting in Configuration_adv.h.

sh4jonny, 2018-02-15:
@thinkyhead @Bob-the-Kuhn i fully confirm that! as you can read on my post at 10 days ago
marlin 2 work fine with a4988 but not with DRV8825. At the same time both drivers work fine in mk4duo.
 so what is default value for MINIMUM_STEPPER_PULSE and what range of values i can try!?
i look now and by default i have uncommented it in mk4duo config
#define MINIMUM_STEPPER_PULSE 0

so i can try later in marlin 2

p3p, 2018-02-15:
The value (microseconds) is driver specific, and can be found in the appropriate datasheets (http://www.ti.com/lit/ds/symlink/drv8825.pdf Section 7.6), DRV8825 has much longer minimum pulse duration than others at 1.9us so setting MINIMUM_STEPPER_PULSE to 2 should be fine.

smoki3, 2018-02-15:
I will try it tommorow.

sh4jonny, 2018-02-15:
@smoki3 i try it now and it work good!

sh4jonny, 2018-02-15:
@thinkyhead @Bob-the-Kuhn @p3p thank you all!!! for your support and your work!

closed 2018-02-23 by thinkyhead