MarlinFirmware/Marlin #9256: Ender-4 Printer config incorrect

lucazader, 2018-01-20:
Hi All,
I have just got an ender-4 printer. Working great so far, just wanting to put standard marlin on it.

According to the issues: 
#9160 and #9188 
that were merged 5 days ago, Marlin now supports my printer.

However it looks like the board that comes with the ender-4 is different than the board used by the ender-2. Not sure if it previously came with a different board.
Picture of board that arrived with my printer:
[https://goo.gl/nUFxhe](url).

The board is based on the ATMega2560, with non integrated drivers and a few other things.

There is a guy (Skorpi08) over on the http://www.3d-druck-community.de forums that seems to have created a config file etc for this board. 
I have created a gist rather than just linking to the complete marlin folder. Let me know if this would be needed and i will upload it.
[https://gist.github.com/lucazader/a6c87e68ef782fe183233e6f8a131e5c](url)

Looking at a diff of the config files there are quite a few. The main one is that the Ender-4 is a corexy printer.

Anyway just thought I would bring this up here even though it is closed and merged. Rather than have someone use this config without checking.

Please let me know if you should make a PR containing the new/correct config for this printer.

thinkyhead, 2018-02-14:
It now has `BOARD_CREALITY_ENDER`. If the config is now working then this can be closed.

lucazader, 2018-02-16:
Hi @thinkyhead 

Thanks for the reply.
As i mentioned in the code, the config for BOARD_CREALITY_ENDER is incorrect if intended to be used with a Creality Ender-4

This config assumes that the ender 2 and the ender 4 printers:
1) use the same PCB
2) use the same mechanics

This is not the case.
The PCB is different, with replaceable stepper drivers and based of a ATMega2560 which is different from the board that comes with the Ender 2.

The printer uses an H-Bot printer style of mechanics.
Which from much understanding should mean that one of the `#define COREXY` or `#define COREYX` defines enabled in the example `Configuration.h`

In the mean time i would suggest that this config should be changed to not refer to the `Ender 4` at all, and separately i will create a PR to add the `Ender 4` as a separate printer once I have tested out the configuration on my own printer.

Thanks,
Lucas

thinkyhead, 2018-02-24:
Awesome. Looking very much forward to it! Version 1.1.9 is due to be released in a couple of days.

thinkyhead, 2018-02-25:
Bitmaps are mentioned in the README. I assume they are the same as those we already have for the Ender 2.

thinkyhead, 2018-02-25:
Can you confirm that the `SD_DETECT_INVERTED` setting is correct? It should be disabled for most displays.

closed 2018-02-25 by thinkyhead

D-side, 2018-03-31:
Probably worth mentioning somewhere that Ender-4 and CR-8 have exactly the same mainboard.
I'm not so sure that the configuration is different or not. I did not manage to get the configuration from this PR to compile, but I have CR-8's original firmware backed up.

thinkyhead, 2018-03-31:
> I did not manage to get the configuration from this PR to compile

What error did you get? Testing with `bugfix-1.1.x`?

D-side, 2018-04-01:
@thinkyhead yes, it was the base branch of the 1.1 PR, `bugfix-1.1.x`.

So just to be safe of my bolted-on mods in an attempt to fix this, I properly cloned the repo (was working with a ZIP until now), copied the configs:
```bash
cp "Marlin/example_configurations/Creality/Ender-4/*" "Marlin"
```
And hit "build" in Platformio. Here's the output (minus some useless noise):

```
PLATFORM: Atmel AVR > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
SYSTEM: ATMEGA2560 16MHz 8KB RAM (248KB Flash)

Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib> v1.19.1
|-- <TMC2130Stepper> v2.2.1
|   |-- <SPI> v1.0
|-- <Adafruit NeoPixel> v1.1.6
|-- <LiquidTWI2> #30aa480
|   |-- <Wire> v1.0
|-- <TMC26XStepper> #c1921b4
|   |-- <SPI> v1.0
|-- <Wire> v1.0
|-- <SoftwareSerial> v1.0
|-- <LiquidCrystal> v1.0.5
|-- <SPI> v1.0
Compiling .pioenvs\megaatmega2560\src\G26_Mesh_Validation_Tool.cpp.o
Compiling .pioenvs\megaatmega2560\src\I2CPositionEncoder.cpp.o
Compiling .pioenvs\megaatmega2560\src\M100_Free_Mem_Chk.cpp.o
Compiling .pioenvs\megaatmega2560\src\Marlin.ino.cpp.o
In file included from Marlin\MarlinConfig.h:33:0,
from Marlin\G26_Mesh_Validation_Tool.cpp:27:
Marlin\pins.h:203:53: fatal error: pins_CREALITY_ENDER.h: No such file or directory

#include "pins_CREALITY_ENDER.h"    // ATmega1284P
^
In file included from Marlin\MarlinConfig.h:33:0,
from Marlin\M100_Free_Mem_Chk.cpp:49:
Marlin\pins.h:203:53: fatal error: pins_CREALITY_ENDER.h: No such file or directory
  
#include "pins_CREALITY_ENDER.h"    // ATmega1284P
^
In file included from Marlin\MarlinConfig.h:33:0,
from Marlin\I2CPositionEncoder.cpp:31:
Marlin\pins.h:203:53: fatal error: pins_CREALITY_ENDER.h: No such file or directory

#include "pins_CREALITY_ENDER.h"    // ATmega1284P
^
compilation terminated.
compilation terminated.
compilation terminated.
*** [.pioenvs\megaatmega2560\src\M100_Free_Mem_Chk.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\G26_Mesh_Validation_Tool.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\I2CPositionEncoder.cpp.o] Error 1
 [ERROR] Took 3.26 seconds
```

D-side, 2018-04-01:
Okay, now having slept for a good while I actually got it to compile 😅 See #10271

k0d3x8its, 2018-06-20:
Can you provide the corrected code that compiles? This is my first 3D printer kit, and I can’t seem to find any information about the Creality Ender 4

thinkyhead, 2018-06-20:
@k0d3x8its — Marlin `bugfix-1.1.x` and `bugfix-2.0.x` both include an example configuration for Ender-4.

willysmoose59, 2018-07-29:
Is there anywhere else to get this config file? I've been waiting to get confirmed on the that site and still nothing. Replaced my hotend on my cr-8 clone and need to load the firmware corrections.

D-side, 2018-07-29:
@willysmoose59 as mentioned above, it's available in `bugfix-*`  in this repository.
Here are the ones in `bugfix-1.1.x` for instance: https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x/Marlin/example_configurations/Creality/CR-8

willysmoose59, 2018-07-30:
@D-side Thanks for the info. I see the corrected information in the fix files but the board listed is not in the boards.h file. Is there an updated boards.h out there with the correct info?

AnHardt, 2018-07-30:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/boards.h#L81