MarlinFirmware/Marlin #9257: UBL bed leveling routine shows large discrepancies between runs

Jens5395, 2018-01-20:
I am running Marlin 1.1.8 on a Creality 10S printer with a BLTouch probe.
 The probe is set up to report as a z max endstop.
I control the printer via OctoPrint

The problem is totally repeatable .....

If I run a G29 P1 bed sensing routine (UBL is enabled) I get a map of reported heights.
If I run the same routine multiple times with no other printing action in between, I get extremely bad repeatability. On the other hand, if I run an M48 test to check on the accuracy of the probe it shows a good repeatability- deviation seems to max out at 0.015 mm which is about what I was expecting. The G29 command gives me deviations for the same spot on different runs of up to 0.15 mm !!!!
Measurements with a dial indicator attached to the head and doing a G28 shows virtually no deviation of head height between different runs.
I have run tests after erasing the eeprom and then running multiple G29 patterns and they all have poor repeatability save one run that had identical data as the previous run (this is highly unusual).
I have run M48 test runs with multiple legs of movements between probings and again, repeatability is excellent. 

- Expected behaviour: repeatability on multiple g29 p1 probing runs should be roughly equal to repeatability achieved in M48 probe accuracy tests.
- Actual behaviour: errors are a magnitude higher than what the m48 tests show is possible
- Steps to reproduce:
  - Do this ... run g29 p1 several times in a row and then compare the readings at the same points in the result matrix
  - Do that ... none of that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
Not attached as the issue happens without changing anything between m48 and g29 runs. I can supply the files if desired. 
I can supply a listing of the heights reported on several G29 p1 runs if desired

Jens5395, 2018-01-20:
This is the output of three successive G29 P1 runs. G29 was set up for double probing each point with the second probing happening at half speed. No bed or tip heat was used, nothing was printed in between the probing runs. I have no clue what other tests I could run in order to find out why this is happening.

Probing run # 1

-0.197   -0.083   -0.159   -0.132   -0.175   -0.137   -0.192   -0.159 -0.235   -0.218
-0.063   -0.068   -0.086   -0.112   -0.112   -0.186   -0.139   -0.217 -0.147   -0.152
-0.147   -0.117   -0.054   -0.120   -0.113   -0.148   -0.149   -0.216 -0.200   -0.171
-0.039   -0.083   -0.129   -0.050   -0.091   -0.109   -0.131   -0.237 -0.198   -0.205
-0.113   -0.143   -0.125   -0.064   -0.104  [-0.126]  -0.144   -0.191 -0.258   -0.297
-0.179   -0.144   -0.139   -0.077   -0.112   -0.144   -0.152   -0.201 -0.220   -0.233
-0.106   -0.058   -0.102   -0.040   -0.076   -0.091   -0.104   -0.207 -0.204   -0.175
-0.081   -0.099   -0.142   -0.139   -0.095   -0.125   -0.129   -0.199 -0.243   -0.242
-0.152   -0.136   -0.046   -0.057   -0.106   -0.135   -0.143   -0.152 -0.163   -0.249
-0.133   -0.130   -0.157   -0.121   -0.058   -0.093   -0.127   -0.237 -0.116   -0.146

Probing run #2

-0.053   -0.023   -0.047    0.058    0.073    0.079    0.062    0.085 -0.033    0.079
 0.046    0.009    0.068    0.053    0.081    0.094    0.090    0.044  0.129    0.091
-0.082    0.061    0.083    0.118    0.083    0.064    0.076    0.061 0.092    0.124
 0.039   -0.007    0.083    0.064    0.072    0.063    0.104    0.021  0.096    0.100
 0.019   -0.051    0.005    0.101    0.069  [ 0.077]   0.048    0.025  0.053    0.058
 0.016   -0.072    0.054    0.086    0.083    0.072    0.045    0.015  0.076    0.077
 0.050    0.006    0.108    0.127    0.109    0.115    0.115    0.031  0.069    0.117
-0.053    0.042   -0.005    0.103    0.074    0.078    0.065    0.050 0.067    0.050
-0.061    0.066    0.088    0.085    0.079    0.101    0.094    0.083 0.076    0.074
-0.044    0.010    0.056    0.006    0.114    0.096    0.071    0.042 0.083    0.104

Probing run #3

-0.089   -0.019   -0.076    0.061    0.065    0.072    0.065    0.090 -0.052   -0.001
-0.068    0.008    0.083    0.101    0.094    0.105    0.118    0.054 0.124    0.029
 0.024    0.050    0.098    0.123    0.090    0.090    0.102    0.106  0.092    0.118
 0.027    0.051    0.089    0.056    0.084    0.109    0.108    0.026  0.110    0.093
-0.033   -0.006    0.057    0.097    0.089  [ 0.097]   0.103    0.017 0.037    0.039
-0.099   -0.002    0.077    0.092    0.079    0.079    0.101    0.038 0.058    0.071
-0.066    0.065    0.113    0.109    0.111    0.126    0.129    0.053 0.072    0.105
-0.062    0.024   -0.018    0.105    0.083    0.079    0.086    0.042 0.086    0.077
-0.071    0.067    0.093    0.123    0.024   -0.003    0.105    0.136 0.097    0.090
-0.030    0.009    0.058    0.098    0.110    0.092    0.072    0.050 0.084    0.082

Looking this over, it looks like the same spot can be measured up to almost 0.3 mm different between different runs

Roxy-3D, 2018-01-20:
Where are you homing the Z-Axis ?    It is best to do that in the center of the bed if possible.

When you do a G28 followed by a G29 P1,    Are the mesh points close to the home location very close to 0.000 mm ?

Jens5395, 2018-01-20:
On 2018-01-20 07:57 AM, Roxy-3D wrote:
> Where are you homing the Z-Axis ? It is best to do that in the center of 
> the bed if possible.

It's difficult to say when all I did homing - I homed a lot. Of course 
every R29 run does a home to begin with. The head goes to bed center 
after a homing.
I did run ten runs with a dial indicator mounted to the head. I wanted 
to verify if the home switch might not always trigger at the same point. 
The sequence was to issue the G28, the head would travel to 0/0, the 
head would then travel to the center of the bed and drop to Z 0 at which 
time I took the reading on the dial indicator. 9 of the 10 readings were 
identical as far as I could tell with one reading higher by maybe 0.0001 
mm (the dial indicator had a resolution of 0.001 per division)

> When you do a G28 followed by a G29 P1, Are the mesh points close to the 
> home location very close to 0.000 mm ?

The mesh points are distributed throughout the bed area (300 by 300) 
with a setback of 10 mm from three sides and maybe 25 mm or so on the 
right side.

Jens5395, 2018-01-20:
Also... It might be helpful to do M48's in several places on the bed and turn on the Schizoid mode to force some X & Y movement on each probe. Is it possible the X & Y movement is causing repeat-ability problems?

Did that .... don't know what 'Schizoid' mode is but I did do M48's with the L5 option to do 5 legs of movements before returning to the original probe spot to do another probing - results show excellent repeatability. The same repeatability as seen when not moving between probings. That was my attempt to hopefully eliminate the possibility that movement introduced an error. 

Just as and FYI: UBL uses the exact same probing system that the other bed leveling systems use. If you switch to a different bed leveling system, do you see inaccurate probing? (My guess is you will.)

I have not tried switching to different bed levelling but if it helps I am more than happy to give that a go.

jeno007, 2018-01-20:
I'm using ABL Bilinear, and in the past run the same repeatability tests with the same results as Jens5395. If I remember well, If there is a G28 Z before every G29, than the repeatability is much better. But it should be checked.
G28 Z
G29
G29
 is NOT the same as
G28 Z
G29
G28 Z
G29

Roxy-3D, 2018-01-20:
> don't know what 'Schizoid' mode is but I did do M48's with the L5 option to do 5 legs of movements before returning to the original probe spot to do another probing - results show excellent [...]

I was thinking something like M48 S L 7 P 10

Jens5395, 2018-01-20:
On 2018-01-20 12:54 PM, Roxy-3D wrote:
> don't know what 'Schizoid' mode is but I did do M48's with the L5
>     option to do 5 legs of movements before returning to the original
>     probe spot to do another probing - results show [...]

m48 s l7 p10

  Mean: 1.735000 Min: 1.725 Max: 1.742 Range: 0.018
  Standard Deviation: 0.005916
  X:150.00 Y:150.00 Z:10.00 E:0.00 Count X:12000 Y:12000 Z:4000

No idea what the 's' does as it doesn't seem documented yet but the 
above is the output performed in the middle of the bed.

Again, this result seems to confirm the accuracy of the probe.

Roxy-3D, 2018-01-20:
The S causes a star shaped back and forth pattern of the nozzle.   You have a good standard deviation.

Jens5395, 2018-01-20:
On 2018-01-20 02:29 PM, Roxy-3D wrote:
> The S causes a star shaped back and forth pattern of the nozzle. You 
> have a good standard deviation.
> 

The plot thickens ...
I recompiled with Bilateral linear leveling and removed the fade height 
directive. Here are the result of 4 seperate runs. G28 was done prior to 
run 1 and run 4 only.

I haven't done a full analysis but it would seem that maximum deviation 
is about 0.04 mm which seems a bit higher than M48 codes would suggest 
but a whole lot better than the same routine under UBL.

????????????????????

Jens5395, 2018-01-20:
My last posting should have referred to AUTO_BED_LEVELING_BILINEAR 
rather than Bilateral linear leveling - sorry :(

Jens5395, 2018-01-21:
I have given up on bed leveling. I probably spent well in excess of 50 
hours to figure things out without success, VERY frustrating.
I have been unable to wrap my mind around how z offset and z probe 
offset affect the generation of the mesh and how those values affect the 
final z position when something is printed with the mesh bed active.

I am going back to a 4 corner manual bed leveling until there is 
sufficient documentation that explains to us mere mortals how everything 
interacts.

I got close .... but just had to pull the plug. The concept is great but 
I do not think it is ready for prime time.

As an additional comment, my last test had me switch from bilinear back 
to ubl and run two bed measuring routines - the two sets of numbers were 
fairly close which leads me to think that maybe the inaccuracies seen 
earlier were my fault somehow.

It seems that the mesh that is generated is dependent on at least z home 
offset which means any change here will invalidate any numbers that are 
stored. This seems to me to be a fundamental flaw but of course I have 
no idea if there is a valid reason for that interaction. IMHO, the mesh 
that is measured should be a standalone set of numbers that z home 
offset can be applied to dynamically. I have yet to figure out exactly 
why there is a z probe offset parameter (it might be important if the 
probe is used to determine the Z home position but in my case I have a Z 
home switch .... rather than fighting with that, this should be documented)

Thank you for an exciting product and I a looking forward to see how 
this project will blossom in the future.

Roxy-3D, 2018-01-22:
>  I have yet to figure out exactly why there is a z probe offset parameter (it might be important if the
probe is used to determine the Z home position but in my case I have a Z home switch .... 

Really???

Festivejelly, 2018-01-23:
Make sure your x axis isnt twisted. Also people with CR-10's for example will get different readings between probing because the axis can end up dropping a bit between runs. Theres various mods to fix this.

One thing it could be is stepper resolution and skipped steps when moving about. So if I remember correctly on a standard setup your Z can raise 0.04mm on full steps. But if your offset isnt a multiple of that, then you could end up getting some skipped steps depending on where the probing takes place. I could be spouting nonsense here but its just a theory.

closed 2018-02-14 by thinkyhead