MarlinFirmware/Marlin #9260: [Question] Info about disable_all_heaters method

GMagician, 2018-01-20:
inside the function there is:
  HOTEND_LOOP() setTargetHotend(0, e, 18);
  setTargetBed(0);

and then call to DISABLE_HEATER(NR) that do almost the same things. there is a reason for this duplicated code?

AnHardt, 2018-01-20:
The one is setting the target temperature. The other sets the pins directly for the case the regulation does not work anymore.

GMagician, 2018-01-20:
@AnHardt thats not completly true. Latter also set target temperature so first should be a duplicate

thinkyhead, 2018-01-20:
I was wondering about it myself. If both heater disables are in fact needed then it should be explained in a comment so that future developers will not be confused by this duplication.

thinkyhead, 2018-01-20:
The original code contained only the direct heater disable code. The new loop at the top was added at the top by @Bernhard here [in this commit](https://github.com/MarlinFirmware/Marlin/commit/415aadf704ff545de2db90ffaf2fc351af78331a#diff-7e90b29c765c3047c7dc2aa8eff0856cR546).