MarlinFirmware/Marlin #9269: 2.0.x compile error mega 2560

swap81, 2018-01-20:
hello I use 1.1.6 and everything is going well.
Now I have 2.0 loaded and filled but not compile possible.
first filled out the most necessary for testing
, use arduino and ramps1.4, chimera 2 nozzle, bed

-------------------------------------------------- ----------
### Compile Error
Arduino: 1.8.5 (Windows 10), Board: "Arduino / Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

fork / exec C: \ Program Files (x86) \ Arduino \ hardware \ tools \ avr / bin / avr-gcc.exe: The filename or extension is too long.
Compile error for board Arduino / Genuino Mega or Mega 2560.

what can I do?

hallo ich nutze 1.1.6 und alles läuft gut.
jetzt habe ich 2.0 geladen und ausgefüllt abr kompelieren nicht möglich.
erst mal das nötigste zum testen ausgefüllt
. benutze arduino und ramps1.4 ,chimera 2 nozzle , bed

------------------------------------------------------------
### Compile Error
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

was kann ich tun?

Bob-the-Kuhn, 2018-01-20:
Use the Arduino 1.9 beta.  It has that problem fixed.

oakdesign, 2018-01-21:
Alternativ a setup with VSCode or Atom and Platformio IDE works as well instead of using Arduino IDE

swap81, 2018-01-22:
thanks I will first test the 1.9 beta vers.

Many Thanks

Vielen dank
ich werde zu erst die 1.9 beta vers.  testen 

Vielen dank

swap81, 2018-01-22:
ok that worked!
now e0 heats thermo3 24c
but e1 thermo 3 gets an error in heating when repeating!
bed thermo 1 on printer display corekt shown with 23c. at repetier with 0c and error if I want to heat

swap81, 2018-01-22:
what do i have to do in boards h and pins ramps h?
or is it another problem

swap81, 2018-01-22:
I'll explain where I want to go.
xy core d-bot
chemera hotend
nozzle 1 1.75 run out and width sensor
nozzle 2 3mm
e0 and e1 different motors
bedleveling biliniar or ubl
would like to upload the file but irgentwie does not work ..

closed 2018-02-14 by thinkyhead