MarlinFirmware/Marlin #9271: [FR] Full support for RepRapPro Sanguinololou (yes that old!)

indazoo, 2018-01-20:
Hi everybody

I have this old Huxley and use it every now and then. Well I wanted to compile the new firmware since you guys sucessfully released a final version. A big "Thank You" for that!

The Problem:
Normally on Sanguino there is a 4Pin Connector for the heated bed which is switched on/off (MOSFET) via pin 12 (Arduino Pin). So, you can connect the heated bed directly to this connector. This is correctly in current Marlin file "pins_SANGUINOLOLU_11.h" line 90, section "Heaters / Fans".
But RepRapPro used for the huxley a heated bed with on-board MOSFET (for whatever reason). So, from the Sanguino board only the on/off signal went to heated bed. The wiring instructions said one should use the pin "D10" (Arduino pin 10) on the extension header of the Sanguino board.
So if one compiles Marlin with standard "pins_SANGUINOLOLU_11.h" for the RepRapPro, or maybe other printers having the MOSFET on the heated bed, the heated bed will not switch on.

Yes, it is possible to adjust the pin in "pins_SANGUINOLOLU_11.h". This works, but all reprappro users must make that and find the problem first! So, a bit annoying.

Yes, one could find pin 12 on the board and solder the cable there. Maybe that would work too, but then all users must know how to solder and find the problem. Well...

It would be much more convenient for the users to have a "define" in "configuration.h" which automatically adjusts the correct pin in "pins_SANGUINOLOLU_11.h". 
#define I_HAVE_AN_OLD_HUXLEY_WITH_MOSFET_HEATEDBED

What do you think. "Naa, too old stuff?" , "Easy enough that pin change, don't complain, we do not change the code for that" or "holy s... this must be fixed" or? 

Thank you 

P.S. I have the configuration.h for RepRapPro Huxley ready. Do you think it will be included as "example_configuration"? Is there a test procedure?

thinkyhead, 2018-02-14:
This can be done as a board/config combination. The very small boards file will simply redefine the pin, or set a flag for `pins_SANGUINOLOLU_11.h`, and the config file would have this board as its `MOTHERBOARD` value.

So I suppose `pins_HUXLEY.h` and `BOARD_HUXLEY` are the logical choice.

boelle, 2019-02-20:
could this one not be closed since the PR is merged?

indazoo, 2019-02-20:
Yes, I see the Huxley file in the configurations in the current release. 
Thank you all!

closed 2019-07-21 by boelle