MarlinFirmware/Marlin #9274: Pressing encoder wheel button defaults to top of page

jordathan, 2018-01-21:
Feature request. In Marlin 1.0 on a LCD smart controller when advancing in a setting menu, after going back to previous submenu, it would default and put your cursor(highlighted setting) on either control, main, or info screen. But after updating if I press button on say main, it will put the highlighted choose on control instead of info screen for easier menu jumping. Was just wondering where I might find this setting it what it might be called. I dug thru all of marlin but didn't know what I was looking for.thanks

thinkyhead, 2018-02-14:
When going "back" or "up" in the menu tree, Marlin now remembers where you were. This behavior can be changed by this modification:
```cpp
void lcd_save_previous_screen() {
  if (screen_history_depth < COUNT(screen_history)) {
    screen_history[screen_history_depth].menu_function = currentScreen;
-   screen_history[screen_history_depth].encoder_position = encoderPosition;
+   screen_history[screen_history_depth].encoder_position = 0;
    ++screen_history_depth;
  }
}
```

jordathan, 2018-02-14:
Thanks a lot Scott. I'll try to implement this tonight

closed 2018-02-23 by thinkyhead