MarlinFirmware/Marlin #9276: [1.1.8]Lcd Bed Leveling does not seem to work (mesh bed levelling)

Its-my-right, 2018-01-21:
### Bug Report
- Description: Lcd Bed Leveling does not seem to work (mesh_bed_levelling)
- Expected behaviour: Setting all calibration points (9/9 points) to +2mm should up the nozzle by 2mm when print start
- Actual behaviour: print start at height 0.1 (first layer heigt) and does not seem to take into account the lcd_bel_levleling settings configured
- Steps to reproduce:
  - calibrate all points to +2mm when doing mesh bed levelling via lcd_bed_levelling
  - save settings
  - print something

You can find attached the config files.

Could you please tell me if it is a real bug or if I have something not configured properly ?

Thank you !

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1649724/Marlin.zip)

Maesto, 2018-01-21:
(https://github.com/MarlinFirmware/Marlin/files/1650124/Configuration.h.tar.gz)
I too can't seem to get bed leveling working on my Delta. I already checked the probe offsets n times, and the measured bed height makes sense. But after leveling and testing with a sheet of paper it don't seems to care.

I testes linear, bilinear as well as ulb nothing seems to have a effect. I'm currently on the 1.1.x branch.

[Configuration.h.tar.gz]

Its-my-right, 2018-01-22:
Ok, so after a few more searches, I found out I had to add "M420 S1" after the G28 commands added by my slicer at the beginning of each gcode file.

Maybe this could be add to the wiki pages talking about bed leveling ?

closed 2018-01-22 by Its-my-right

ReadieS, 2018-01-22:
M420 S1 sorted the problem for you? I am having a similar issue, I added that to my startup Gcode a while ago but it still seems to ignore it! are you using the g29 command in your gcode as well and if so before or after m420 s1? Are you using the standard mesh bed leveling or the ubl system? Thanks for you help I have been struggling with this for a few days now and couldn't find anyone else having the issue!

Maesto, 2018-01-22:
My Start Gcode currently looks like this:
```
G28 ; Home to top 3 endstops
M420 S1 ; Enable Bed leveling
M42 P45 S255 ; Lights On
M190 R{material_bed_temperature} ; Heat Bed & Wait
M109 S{material_print_temperature} ;Heat Hotend & Wait
```

But neither mesh bed leveling nor ULB seems to work.

Its-my-right, 2018-01-22:
How do you calibrate your leveling ? (lcd ?) do you save the mesh in eeprom ?

Maesto, 2018-01-22:
I calibrated ULB via the LCD and mesh via Terminal. I basically followed this documentation for the mesh leveling: http://marlinfw.org/docs/features/auto_bed_leveling.html
```
G28
G29
M500
```
But even directly after G29 manual checking shows that it doesn't honor its measured values.

Its-my-right, 2018-01-22:
I can't help you sorry :/

ReadieS, 2018-01-23:
Thanks Its-my-right for reassuring me i was on the correct path!
I have got mine working now.

Maestro, I had a bit of trouble saving the calibration the method that worked for me was

In the firmware enable the following ensuring where there are multiple options you only have one selected:
#define EEPROM_SETTINGS
#define LCD_BED_LEVELING
#define MESH_BED_LEVELING
#define PROBE_MANUALLY

Now issue these two commands (I like to do all commands through a terminal on the computer and you can see it has registered in the marlin response) :
M502

This resets eeprom to firmware defaults (this assumes you have eeprom commands turned on in firmware)

M500

This saves the defaults (be careful if you have saved other settings to your eeprom this will wipe them)

Now level the bed using the LCD ( I use the standard mesh bed levelling not the UBL) and you want to be able to be able to just feel positive pressure on a piece of paper at each probe point.
Once it says you have finished and the print head has moved back to home position enter;

M500

To save the mesh to eeprom

Now try entering M240 S1 and it should print out saying bed leveling: on and your probe points from memory

Now in your dafault startup code after the g28 command (some startup codes such as Cura have 2 one for x y and one for z) add the command

M240 S1

Make sure you use capitals for the letters ( it made a difference on mine!)
My startup gcode as an example;

G21 ;metric values
G90 ;absolute positioning
G10 ; retract filament
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 ; Set Home Coordinates
M420 S1 ; Turn on the Eeprom Bed Mesh
G0 X0 Y0 F9000 ; Go to front
G0 Z0.2 ; Drop to bed
G92 E0 ; zero the extruded length
G1 X40 E25 F500 ; Extrude 25mm of filament in a 4cm line
G92 E0 ; zero the extruded length
G1 E-1 F500 ; Retract a little
G1 X80 F4000 ; Quickly wipe away from the filament line
M117 Printing...

It should now be working!
Sorry if that was a bit back to basics for you, I just wanted to cover all bases to try and help you get it working.

Good Luck.

kermur, 2018-06-24:
You have a typo there. 

""
Now in your dafault startup code after the g28 command (some startup codes such as Cura have 2 one for x y and one for z) add the command

M240 S1
""

I believe you mean 'M420 S1' as we see in you startup gcode.

ReadieS, 2018-06-24:
Thanks Kermur. I have adjusted it, I wrote that in quite a rush!

juanggrados, 2018-09-25:
Thanks a lot. Worked on a Anet A8 on Marlin 1.1.9

arran2308, 2018-12-30:
Thanks a million - worked great on my Anet A8!! Really Appreciate the help!