MarlinFirmware/Marlin #9279: [2.x] Bug Report - Buzzer on whilst booting and beyond Re-Arm

psavva, 2018-01-21:
### Bug Report
- Description: The buzzer starts to sound on the RRDFG LCD when the printer starts, and it keeps on sounding until I physically press the knob on the lcd, or issue an M300
- Expected behaviour: The Buzzer should not sound during and after bootup, it should remain silent, unless when selecting menu items, etc
- Actual behaviour: Buzzer sounds during bootup and after...
- Steps to reproduce:
  - Startup the printer power on thr Re-Arm
  - Buzzer Sounds...

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1649929/Config.zip)

p3p, 2018-01-21:
Well this can't be a common issue or I'm pretty sure more people would have complained.. standard wiring to the RRDFG?

psavva, 2018-01-21:
My RRDFG came with the plastic mounts reversed.  
I shaved the key and reversed the connector.
The LCD and SD works perfectly otherwise.

p3p, 2018-01-21:
Compiled with your config other than the TMC2130 setting and the buzzer doesn't sound, does disabling that fix it for you? I assume you're using the TMC2130 PR, I don't have time right now to get that setup but will later if disabling it does fix it for you.

psavva, 2018-01-21:
I am using the TMC2130 PR.
I will compile and disable the TMC2130 and see if it solves it....

Grogyan, 2018-01-21:
@psavva quick simple test, is to unplug the the plug for the SD card.  This will isolate the issue down to whether the soft SPI? is causing the issue or not.

There is quite a bit work going on to rehash the SPI into a separate library I think.

vilsed, 2018-01-21:
I had this problem immediately after connecting the RRDFGLCD after getting TMC's to run. P1_30 pin is also used for RRDGLCD beeper. Get's annoying quite quickly. Moved Z_CS to P1_15

teemuatlut, 2018-01-21:
@vilsed Heh, I was just about to link to your comment because I remembered hearing about an issue like this.
I'll probably need to change all the CS pins to the ETH header as well.

p3p, 2018-01-21:
was about to say that

```
#define Z_CS_PIN           P1_30  // (37) J5
```

That's the beeper pin

vilsed, 2018-01-21:
@teemuatlut Yes, I guess that would make the most sense :)

gavatron3000, 2018-01-28:
my plugs were reversed as well, cheers guys

psavva, 2018-02-04:
Default pins have moved to the ETH Header.  Closing...

closed 2018-02-04 by psavva