MarlinFirmware/Marlin #9280: [bugfix-2.0.x] "KILLED. PRINTER HALTED Please reset" How to unbrick the board?

liberssky, 2018-01-21:
- Description: I've been messing around with Marlin 2.0.x trying to get it working on my AZSMZ MINI + AZSMZ_12864 LCD. I used the MKS SBASE example config, changed the  `#define MOTHERBOARD` and I managed to get Marlin to boot up and LCD to display stuff. Nozzle temperature was showing correctly but bed temp wasn't so I changed the bed thermistor pin config to a digital one (rookie mistake), got it compiled, uploaded to the SD and restarted the board. 
- Now all I get is "KILLED. PRINTER HALTED Please reset", when I'm trying to connect the board to the PC with USB it gets detected, appears as storage for half a second and immediately disappears, seems to be crashing.
- I tried formatting the SD card and copying both Smoothie and corrected Marlin 2.0.x firmware.bin file but it's all for nothing.

How can I unbrick the board?

p3p, 2018-01-21:
Marlin doesn't handle the flashing process from the sdcard (only the usb connection) that is done by the smoothie boot-loader (we don't change it) so the procedure you carried out should have solved the problem.. if the boot-loader is not writing the firmware it may be having issues reading the sdcard after you formatted it?

liberssky, 2018-01-21:
I tried loading a new firmware.bin (a few different ones) before I formatted the card too, so thats probably not it. I will try a different sd card, it can't hurt at this point.

closed 2018-01-21 by liberssky

p3p, 2018-01-21:
I couldn't think of any reason flashing Marlin could damage the boot-loader so I'm very happy that's not the case, hope you get your board working well.

iggut, 2018-06-07:
@ghost any chance I can get the files you changed to get Marlin 2.0 to work with the AZSMZ mini and AZSMZ_12864 LCD please?