MarlinFirmware/Marlin #9285: Skew M852 reporting 0’s

empakoso, 2018-01-22:
Is anyone having difficulty loading the measurements as shown below (while leaving the “define XY_SKEW_FACTOR” set to 0.0?

With the sample below I get M852 reports all 0’s. And continues to print a skewed calibration square.

If I manually calculate and enter the skew factor using the formula provided, it reports the value entered and prints a perfect square.

Could it be that the 0.0 skew factor value takes precedence in the code over the actual measured values?

I am using the latest 1.1.x bugfix (as of 2018-01-20) to report these findings.

***start of config skew changes***
#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 141.05
  #define XY_DIAG_BD 141.46
  #define XY_SIDE_AD 100.89

  // Or, set the default skew factors directly here
  // to override the above measurements:
  #define XY_SKEW_FACTOR 0.0
***end***

mylife4aiurr, 2018-01-22:
https://github.com/MarlinFirmware/Marlin/issues/8843

empakoso, 2018-01-22:
I did read through that thread prior. My only contradiction is that if M852 reports 0’s the resulting calibration square was not using any skew factor (same as before).

I tried the latest bugfix since it seemed to have the fix applied but with the same results.

My only working solution was to leave the measurement (values set to anything, really) and calculating by hand.

It’s a wonderful tool in Marlin, now all my electronic experiment enclosures are printing beautifully.  The lids close perfectly!

I believe the measurements are set to zero by the factor value.  This 0.0 value must be taking precedence over the measurements.

mylife4aiurr, 2018-01-22:
> This 0.0 value must be taking precedence over the measurements.
Yes thats how is suppose to work - 
> // to override the above measurements:
> #define XY_SKEW_FACTOR 0.0

if you want to use your measurements
#define XY_DIAG_AC 141.05
#define XY_DIAG_BD 141.46
#define XY_SIDE_AD 100.89
then comment out this line
> //#define XY_SKEW_FACTOR 0.0

if you hand calculate then put value and dont comment out
> #define XY_SKEW_FACTOR 0.0 <<<----

empakoso, 2018-01-22:
Using 1.1x bugfix build from 20jan2018

My third attempt was to leave my measurements and hand calc skew factor as is, then commented out the skew factor line. Got a sqrt compile error.  How did you work around this?  

For me the skew factor is working. For the next person... the config file and setup of the values could be a lot better.

empakoso, 2018-01-22:
Did I mention that the latest bugfix would report M852 as zeros and would print w/o applying my skew correction measurements.

mylife4aiurr, 2018-01-22:
Latest bugfix fixed it for me.... 
 > Got a sqrt compile error. How did you work around this?
All your questions answered in #8843

empakoso, 2018-01-22:
Ok, I’ll leave this config issue here for others to work through the setup.  In my opinion, load measurements and leave as defaulted (comment out as instructed) should be a more correct method of config setup. Just my $0.02.

mylife4aiurr, 2018-01-22:
I can agree with u there. Even though it's fixed and working correctly in bugfix - its not intuitive, ie instructions could be better. You litterly asked all the same questions and came to similar conclusions

thinkyhead, 2018-01-22:
Should be fixed with the above PRs. Please test and confirm after they're merged.

empakoso, 2018-01-22:
@thinkyhead thank you for the update

closed 2018-01-22 by empakoso

empakoso, 2018-01-29:
Confirmed, all is good.  Skew can be entered by measurements or by calculation.  Enter chosen value and comment out the other.