MarlinFirmware/Marlin #9287: Z axis Stepper motors not working correctly

Bobsta6, 2018-01-22:
I have been searching around the internet for the last 3 days trying to figure this out. My Z axis motors for a pursa-i3 3d printer are not working correctly. I have marlin firmware and using repetier host. I send a command to move the z axis and I get it to move, however I if I send the same command again the motors will sometimes spin the other way. feel like They almost randomly choose which direction they turn.

As I said I have been trouble shooting this for a while now. What I am suspecting is the firmware feedrates and acceleration or some setting is not correct. 

Here is my code:

    //// MOVEMENT SETTINGS
    #define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
    #define HOMING_FEEDRATE {50*60, 50*60, 2*60, 0}  // set the homing speeds (mm/min)
    
    #define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,4000,590}  
    #define DEFAULT_MAX_FEEDRATE          {300, 300, 3, 45}    // (mm/sec)
    #define DEFAULT_MAX_ACCELERATION      {1000,1000,50,500}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
    
    #define DEFAULT_ACCELERATION          1000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
    #define DEFAULT_RETRACT_ACCELERATION  1000   // X, Y, Z and E max acceleration in mm/s^2 for retracts
    
    #define DEFAULT_XYJERK                10   // (mm/sec)
    #define DEFAULT_ZJERK                 0.3     // (mm/sec)
    #define DEFAULT_EJERK                 5.0    // (mm/sec)

I tried swapping the drivers around and the motors will work perfectly on another axis so I don't suspect it to be a driver issue. I have been turning the pots ontop of the drivers to make them work but can't make them to go the same direction i want them to.

I've checked the wires and I almost sure they are wired up correctly. (could be wrong but have checked it over with a multimeter.)

I am new to this and it's my first time building one of these would appreciate any help I can get and and maybe I have over looked something I have tested.
Just really want the axis to move in the direction That I say it to move in.

*****
More details about my setup is: A Robocraze 3D Printer Controller Board RAMPS 1.4 using A4988 stepper motors drivers and my motors are the nema 17 stepper motors. I currently have the two z axis motors wired in parallel but have tried before using series, however the problem of being unable to control the direction of the Z motors still arises (can easily switch back to series). 

currently trying with no load just to get the motors turn in the correct direction when I send a G-code command. I am using Repetier host on ubuntu 14.04.5.

I have also check the endstops and they are working perfectly, so they ain't a problem (I don't think :p)

Thank you,
Bobby

oakdesign, 2018-01-22:
`#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,**4000**,590}`
What is the pitch of your Z rods are you running a M5 leadscrew with 0.8mm pitch?

Bob-the-Kuhn, 2018-01-22:
Problem does not follow the driver.
Problem does not follow the steppers.

I'm thinking that the Z socket has a problem.  Sounds like an open/poor contact.

Try bending the **DIRECTION** lead on the Z driver a little and see if the problem disappears.  It's one of the corner pins.  Sometimes it's called **DIR**.  If your driver's pins aren't labeled then bend all four corner pins a little.

Another option is to move the Z function to the E1 socket.  Replace your **pins_RAMPS.h** file with this one.
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1653412/pins_RAMPS.zip)

FYI - if this really is a hardware problem then it's the second RAMPS hardware problem within a week.  Most unusual.

Bobsta6, 2018-01-23:
@Bob-the-Kuhn Yes the socket  I am now assuming is just broken (not sure what exactly but possibly one of the connections), After using for pins_RAMPS file and changed the motors back to series and connected to the E1 slot I successful got the printer to work!!! Thank you very much Z axis is working as I would expect! I am now calibrating the printer as it definitely needs it.

closed 2018-02-14 by thinkyhead