MarlinFirmware/Marlin #9291: [FR] Is there a way to compile and/or upload without the arduino-IDE

schorsch3000, 2018-01-22:
### Feature Request
#### prolog
At first, this is not a Feature Request, but is it the closes match.
I even don't knof if this is the right place to ask, but i did not found a mailinglist or a chat, so this seems to be my only option, so feel free to point me to the right place for something like this.

#### My real Question
My current setup as an Anet A8 (with optiboot), running Marlin 1.1 connected to a raspberry pi running Octoprint.
My current workflow for flashing new / reconfigured version of marlin is:
 - remove my laptop from its dockingstation
 - walk over to the printer
 - disconnect the printer from the RPI
 - connect the printer to the laptop
 - open the arduino ide and upload
 - reconnect and reloacate everything

I really would like do simplify this workflow, basically i would like to flash the printer from the rpi.
Optional i would like to compile marlin without the arduino-ide.

Is this possible and there some Documentation anywhere how to do this?
I don't need a ready-out-of-the-box solution, i'll happily write some shellscripts to glue everything together, there are 2 things that i don't know how to do:
 - how can i compile marlin without the arduino-ide, or with the arduino-ide, but headless.
 - how would i flash the binary from the step before, is that just a avrdude call with the right params? maybe thay are even in the arduino-ide's log?

Thanks Dirk

oakdesign, 2018-01-22:
Have a look at Platformio CLI which would be able to run of a any Platform including RPi and can perform all the requested task such as compile, upload from command line
http://platformio.org/get-started/cli

schorsch3000, 2018-01-22:
That look great, i'll try out later, thanks!

benlye, 2018-01-22:
This OctoPrint plugin lets you flash compiled firmware hex files to the printer from OctoPrint:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater

oysteinkrog, 2018-01-22:
The current platformio setup works great for exactly this scenario, just make sure to disconnect from the printer in octoprint before you run the pio command to compile and upload.

tcm0116, 2018-01-22:
OctoPrint-FirmwareUpdater is exactly what you want. You can keep your laptop in the dock, build Marlin, and then flash the hex file to your printer from the Pi via OctoPrint.

schorsch3000, 2018-01-22:
So, platformio on my Laptop and FirmwareUpdated in Octoprint bundles with sum curl in a shellscript seems to be my sweet spot.
I'll give it a try and will report back.
Thanks so long for all the hints!

benlye, 2018-01-22:
> So, firmware-updater doesn't work for me, the dialog closes, no logline is written.

Did you actually click the big blue 'flash from file' button after selecting your hex file?

schorsch3000, 2018-01-22:
> Did you actually click the big blue 'flash from file' button after selecting your hex file?
In fact: no, i didn't get it.

While trying to reproduce my steps, i'm quite sure that button was disabled all the time, maybe i've missed some setting last time.
I tried again, it works as it should.
But i stay with my solution, less mouse movement included :)

benlye, 2018-01-22:
The button stays disabled until all the requirements are met - if you hover over it while it is disabled you get a tool-tip telling you why it is disabled.  I'm the plug-in author, so I just wanted to make sure it wasn't a bug.

I'm glad it worked in the end, even if you don't use it.