MarlinFirmware/Marlin #9292: Serial Communication- Unfinished Response

ecoe, 2018-01-22:
------------------------------------------------------------
### Bug Report
I am writing Java code to communicate with Marlin in order to create some fully-autonomous computer control. I've figured out that the serial flow control is primarily through the "ok" that Marlin sends. However, right now my controller isn't responding with the full message until queried a second time. For example, sending M119 will yield 
```
Reporting endstop status
x_min: TRIGGERED
y_min: TRIGGERED
```
and then when I send something else like M105, it will respond
```
z_min: TRIGGERED
ok
ok T:-15.00 /0.00 @:0
```
The same seems to be true for any multi-line response, including it's initial M503 upon connecting. It will print the majority of its message, and then pause until I send it something new. 

I suspect this is a problem in my Java implementation, but I've been scouring the web looking for documentation on Marlin's serial protocols and coming up empty. Do I need to send a response to the control board letting it know I received its message? Do I need to set the serial port to a new state?

Sorry if this wasn't supposed to be filed as a bug report; I've been working on fixing this for a week and I didn't know where else to ask.

tcm0116, 2018-01-22:
It sounds like you may have a buffering issue in your Java code. The flow control is exactly as you think: you send a command and Marlin sends "ok" back.

tcm0116, 2018-01-22:
What are you using in your Java code to perform the Serial communications?

ecoe, 2018-01-22:
I was in the middle of responding that I had weeded out buffering as a cause by simply printing to terminal the "input" in my code every time there was data available... when I realized I had defined "input" as a BufferedReader of the stream, not the stream itself. 

I feel very, very, very dumb right now... 

Thank you so very much for your help; you have ended a week-long headache for this one-person dev team.

closed 2018-01-22 by ecoe

tcm0116, 2018-01-22:
Glad you got it figured out!