MarlinFirmware/Marlin #9293: [FR] Ultimaker 3 I2C print head support

burdickjp, 2018-01-22:
I'm looking to use an Ultimaker 3 print head in a machine using RAMPS 1.4.2. It looks like everything in the print head communicates back to the controller over I2C. https://github.com/Ultimaker/Ultimaker3/tree/master/PCB%20files/2012-K%20Print%20head%20PCB
What would be involved in Marlin supporting this?

The alternative would be replacing the PCB in the print head with something which deletes all of the I2C business and routes everything back to the controller, like most every other 3D printer on the market. This doesn't look like all that much work, but it sure would be cool to use the Ultimaker 3 print head as it is.

Grogyan, 2018-01-22:
Ultimaker 3 using I2C for the hot ends is questionable.  As the BUS isn't good for long distance communication over more than 5- 10cm.

Perhaps it is worthwhile letting Ultimaker know about this flaw.
CAN bus is the solution in this scenario.

burdickjp, 2018-01-23:
It works very well.

burdickjp, 2018-01-23:
I sent a message to Daid at Ultimaker asking for the appropriate firmware source code. He replied with this:

> Naturally I can share that. (GPL even requires that we share it ;-))
> Here you go:
> http://software.ultimaker.com/jedi/Marlin-2017.06.21.zip
> Most important bit is the i2c_driver, which is "new" compared to other Marlin code, and allows multiple things to happen on I2C with different priorities. The rest of the I2C head communication is all using that driver.
> The head is nothing special. It's a ADS101X ADC for temperature, a PCA9635 LED IO extender for the fans and leds, a FDC1004 chip for the sensing for leveling, and DS2482/DS2431 for the hotend EEPROM (which isn't essential, used for PrintCore detection)

I'm going to go through the source and see what I can find. If I remember correctly they stripped a lot out of the firmware for the UM3. I think they drip G-code from their embedded linux computer.

burdickjp, 2018-01-23:
I found this prior feature request for much the same thing: https://github.com/MarlinFirmware/Marlin/issues/7742

Difference here is that we have an existing piece of hardware and the necessary code to implement it.
Just the daunting task of smashing Ultimaker code into upstream Marlin. The joys of forks.

thinkyhead, 2018-02-14:
> Most important bit is the i2c_driver, which is "new" compared to other Marlin code, and allows multiple things to happen on I2C with different priorities. The rest of the I2C head communication is all using that driver.

Sounds like something we should look to integrate.

burdickjp, 2019-02-26:
Anything recent with this?
I've got some time away from my hardware projects coming up and would like to get something going for a UM3 head on a RAMPS based system. My options are to draw up and make a new board to drive the hardware directly or to talk to it the way Ultimaker intended.

thinkyhead, 2019-03-06:
No one has submitted code supporting the i2c toolhead so far. We hope that there is enough interest and someone will set aside the time to research, implement, and submit it. All our most active developers are busy with other projects at this time, but we often get submissions for much-in-demand features out of the blue.

Swwils, 2019-03-13:
I renamed a ultimaker print core that was wrongly tagged today via gcode file running M command as per:

http://gr5.org/cores/

`M151 T0 A8 D7800000000004242
M151 T0 A16 D20302E3800000000`

Looks like the ultimaker flavour of marlin is using I2C well in the print head etc.

Definitely a feature to consider upstreaming from their fork. As lots of printers are doing "smarter" heads now and it opens the door to lots of user error prevention features. Yey forks!

boelle, 2019-06-22:
missing label

burdickjp, 2019-08-21:
Can I put a bounty on this?

othermod, 2019-09-11:
I'm giving this a try. I just bought a UM3 print head, and I'm going to start small and just communicate with the chips. The bare minimum will be the ADC to get temps and the LED driver to get fans, then I'll see whether I can get it working in Marlin.

My coding skills are pretty basic, but I'll hopefully be able to copy/paste Ultimaker's code into something usable.

I'll update if/when I succeed.

othermod, 2019-09-16:
I got the wiring all figured out.
The schematics are all here:

https://github.com/othermod/MarlinUltimaker3

othermod, 2019-09-16:
I've got some of the electrical stuff worked out. 

Here is the pinout for the print head wire:
Left to right on this image https://i.imgur.com/QfhB2hM.jpg

Pin 1(blue)	24v
Pin 2(black)	Left Heater-
Pin 3(white)	Left Heater+
Pin 4(gray)	Right Heater-
Pin 5(purple)	Right Heater+

24v pins 1, 3, and 5 are all connected to the same point on the Ultimaker 3 mainboard. Pins 2 and 4 are connected to mosfets and switch to GND to enable the heaters.

Pin 6(green)	GND
Pin 7(yellow)	SDA
Pin 8(orange)	SCL
Pin 9(red)	+5v
Pin 10(brown)	GND

Pins 6 and 10 are connected together. 

All the chips are powered by 5v. There is a 3.3v low-noise voltage regulator that drops the 5v down to a cleaner 3.3v to power a couple of the chip (ADC and bed sensor), but that shouldn't have any effect on what we are doing.

Pins 7 and 8 each need 5v pullup resistors (1.5k on UM3) if your mainboard doesn't have pullups. The pullups are located on the UM3 mainboard, not the printhead board. The printhead has an EM filter on the I2C lines. There is normally an EM filter on the board side too, but communication works fine without having one board-side.

Chips I have tested:
ADS1015 for PT100 print core temp sensors
PCA9635 for LED and fan control

Chips I have not tested yet:
DS2482 (I2C to 1-wire) for left print core
DS2482 (I2C to 1-wire) for right print core
FDC1004DSCT bed sensor

Ultimaker has all the schematics posted here:
https://github.com/Ultimaker/Ultimaker3/tree/master/PCB%20files

I got the temp sensors, fan control, and RGB LED control working with a simple arduino sketch, but getting this into Marlin is way over my head at the moment.

I forked what I believe is the most recent publicly available source for the UM3 board. This might be easier to get up and running temporarily, but even this is proving difficult.
https://github.com/othermod/MarlinUltimaker3

I changed the serial port from 2 to 0, and flashed this to a standard board. Communication works for a moment then is goes silent and doesn't respond to anything. It was killing because of STOP_REASON_SAFETY_TRIGGER, and I think I have that resolved by pulling pin 28 low with a resistor. I'm stuck there at the moment.

I have a lot of testing ahead of me on this. If anyone with skill wants to get involved, I'm happy to flash your code to my board.

thinkyhead, 2019-09-17:
You should be able to do a lot with the `EXPERIMENTAL_I2C_BUS` included in Marlin. I have used it to implement some things in the past. See https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

othermod, 2019-09-17:
Thanks. I also found a great reply from a while back here https://github.com/MarlinFirmware/Marlin/issues/7742#issuecomment-332136054 that was helpful. 

The code below from temperature.cpp appears to be whats getting the onboard ADC reading (from my limited understanding at least), and I assume the ADS1015 needs to go in here in place of the HAL_START_ADC and ACCUMULATE_ADC. I'm searching for a needle in a haystack and I can't tell the difference between a needle and hay yet, but I'm getting there. I'll keep absorbing info until I make it happen.

```
#define ACCUMULATE_ADC(obj) do{ \
    if (!HAL_ADC_READY()) next_sensor_state = adc_sensor_state; \
    else obj.acc += HAL_READ_ADC(); \
  }while(0)

  ADCSensorState next_sensor_state = adc_sensor_state < SensorsReady ? (ADCSensorState)(int(adc_sensor_state) + 1) : StartSampling;

  switch (adc_sensor_state) {

    case SensorsReady: {
      // All sensors have been read. Stay in this state for a few
      // ISRs to save on calls to temp update/checking code below.
      constexpr int8_t extra_loops = MIN_ADC_ISR_LOOPS - (int8_t)SensorsReady;
      static uint8_t delay_count = 0;
      if (extra_loops > 0) {
        if (delay_count == 0) delay_count = extra_loops;  // Init this delay
        if (--delay_count)                                // While delaying...
          next_sensor_state = SensorsReady;               // retain this state (else, next state will be 0)
        break;
      }
      else {
        adc_sensor_state = StartSampling;                 // Fall-through to start sampling
        next_sensor_state = (ADCSensorState)(int(StartSampling) + 1);
      }
    }

    case StartSampling:                                   // Start of sampling loops. Do updates/checks.
      if (++temp_count >= OVERSAMPLENR) {                 // 10 * 16 * 1/(16000000/64/256)  = 164ms.
        temp_count = 0;
        readings_ready();
      }
      break;

    #if HAS_TEMP_ADC_0
      case PrepareTemp_0:
        HAL_START_ADC(TEMP_0_PIN);
        break;
      case MeasureTemp_0:
        ACCUMULATE_ADC(temp_hotend[0]);
        break;
    #endif

```

ulsymn, 2020-02-04:
@othermod Did you fully control the printhead over I2C? Can you provide your latest Arduino sketch?