MarlinFirmware/Marlin #9294: Unicode Static menu bug

karabas, 2018-01-22:
### Bug Report
- Description: ---
Marlin 1.1.16  1.1.8
Cyrillic support is incorrect for static menu items. For LCD2004 ( rrd smart controller)

- Expected behaviour: ---
correct unicode translation as it is for nonstatic menu items.

- Actual behaviour: ---
F.e.
Move axis -> Move X -> Move 10mm -> 
Correctly displaying.
-> Move X: xxxx 
Incorrectly displaying. 
Looks like it forgets about unicode  and displays raw strings, two bytes per char.  

Something wrong with STATIC_MENU functions.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

karabas, 2018-01-23:
Printer Info translates correctly

thinkyhead, 2018-02-11:
I'm unable to see any difference in `lcd_implementation_drawmenu_static` and `lcd_implementation_drawmenu_generic`. They both use `charset_mapper` to put characters to the display.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@karabas Please test with the latest `bugfix-2.0.x`

closed 2019-05-16 by thinkyhead