MarlinFirmware/Marlin #9300: Ultimachine Einsy Rambo display issue

stottsjr, 2018-01-22:
Hi guys, 

I've bought and installed an Einsy board into my printer and the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" that I have on the printer displays garbled characters on the screen.  (See image)

[image]

The display is still functional and I've tried Marilin version 1.1.8 and 1.1.5.  

I've attached the pins file that I'm using.  Hoping it's something simple that I've seemed to have overlooked.

Thanks!

[pins_EINSYRAMBO.txt](https://github.com/MarlinFirmware/Marlin/files/1652615/pins_EINSYRAMBO.txt)

AnHardt, 2018-01-22:
Increase delays.

stottsjr, 2018-01-22:
Ah, I see the MINIRAMBO delays that I missed when I was implementing a EINSYRAMBO configuration, I'll try those when I get home and report back.  

Thanks!

stottsjr, 2018-01-22:
Yep that was it!

Thanks An!

closed 2018-01-22 by stottsjr

kevinmce, 2018-01-30:
Sorry to go off topic. But where did you source the Einsy Rambo? I don't see any in stock.

stottsjr, 2018-01-30:
I got mine from Ultimachine when they announced the release. (I was on their mailing list for it.)

https://ultimachine.com/products/einsy-rambo-1-1?variant=5522483773470