MarlinFirmware/Marlin #9301: ANET A8 Auto bed level Configuration help needed

minorminer59, 2018-01-22:
Desperately need guidance on configuration of the autobed level sensor. I have followed most guides and videos and cannot get this right. 

Goal is to auto home correctly and auto level correctly. Thank you 

My configuration info

X_MIN_POS -33
Y_MIN_POS -10
Z_MIN_POS 0

All _probe_offsets are zero for now
This gets it pretty close to center of the board when I auto home from the lcd. The LCD reads X110 Y110 Z11.3

My probe is front of the extruder 40mm and left 17 

If I enter -17 for the X and -40 Y it almost puts the probe at the center of the board.

If I go back and alter the original X and Y home POS to offset the probe offsets, It gets way off. 

To make it worse when I select autobed level from the LCD it crashes into the board because the probe is off. (Probably need to adjust those margins too.) 

Can anyone help me figure this all out? I’ve gone through all of the previous posts and YouTube videos without any success. 

Here is the Probe holder.
https://www.thingiverse.com/thing:1745261/#comments 

18mm version

benlye, 2018-01-22:
Zip and post your config files, but most likely you need to set:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER -17 // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -40  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0    // Z offset: -below +above  [the nozzle]
```

A word of caution though, don't take the part designer's word for the probe offsets, measure them yourself to confirm them.  If they're out by 1-2mm they will screw up your levelling.

minorminer59, 2018-01-22:
Benlye thank you for the quick reply. I will upload them when I get home from work. 

I did try this and the auto home was very far off. But I’ll upload the config.h file later.
Thanks again

benlye, 2018-01-22:
With probe offsets set and `Z_SAFE_HOMING` enabled the printer will home with the probe in the middle of the bed.  This is intentional, and best left that way.

minorminer59, 2018-01-22:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1653572/Configuration.zip)

brisma, 2018-01-23:
Hi minorminer59,
try to follow this video, is a good starting point and everything will work correctyl :)
https://www.youtube.com/watch?v=WWDkZtWwd6I

And remember to calibrate the Probe/Z-Offset:
https://www.youtube.com/watch?v=y_1Kg45APko

minorminer59, 2018-01-29:
These videos helped out a ton. Thank you. However, my Auto home and auto probing are still skittles off. 

How do I step the sensor to the center and then store those settings so when I autohome it’s perfectly centered? 

Second, how do I keep the sensor on the bed when I auto level? 

Thank you again.

ManuelMcLure, 2018-01-30:
A couple of things you might want to do:

- Define Z_SAFE_HOMING in your configuration. What this will do is home X and Y first, and then move the probe to the center of the bed before homing Z. This makes sure that the probe is always over the bed when leveling. This assumes that your X_MIN_POS, Y_MIN_POS, X_BED_SIZE, Y_BED_SIZE, X_PROBE_OFFSET_FROM_EXTRUDER, and Y_PROBE_OFFSET_FROM_EXTRUDER settings are correct. If they are, the probe should automagically end up in the center of your bed.
- Define PROBING_HEATERS_OFF if you're using an inductive sensor - the current flowing through a heated bed can cause the sensor to be erratic if the heater is on when probing.
- Heat your bed before doing your bed level. Beds can flex with temperature changes and leveling with the bed cold can cause your measured bed level to be off. My start gcode looks like this (I'm using Bilinear leveling with a 3x3 grid):

M851 Z-0.75  ; set Z probe offset
M190 S65 ; start heating bed and wait until it hits 65C
G28 ; home
G29 ; bed level
M109 S210 ; start heating nozzle and wait until it hits 210C

Note that the M190 and M109 settings are actually defined using the slicer's parameters. On Cura:

M190 S{material_bed_temperature_layer_0}
M109 S{material_temperature_layer_0}

and on Slic3r:

M190 S[first_layer_bed_temperature]
M109 S[first_layer_temperature]

closed 2018-02-14 by thinkyhead