MarlinFirmware/Marlin #9303: Bed level seems to ignore Zero-offset

FlailAway, 2018-01-22:
### Bug Report
I have a Anycubic Linear Plus.
I have an E3D and fans mounted on a plate that bolts to the effector.
I have a Dial indicator on a plate that **can** bolt to the effector after removing the E3D plate.
I have a sensing probe on a plate that **can** bolt to the effector after removing the E3D plate.

I mounted the Probe-plate, set the zero-offset to 0.00 and used G28, G33, G29 and plugged all the results into the configuration.h and saved with M500, M502, M501.

I put the E3D plate back in then manually checked nozzle-bed zero and the offset was -0.31mm. I plugged that into confiuration.h and saved,

I then put the dial indicator in to see what's what and using "M420 S1" in the gcode file ran a part-file.

This showed about 0.32mm too high. I manually changed the offset to 0.62mm, ran the part file and again got the same results as above. The indicator showed that the "new" zero offset was not being used.

Now that may have been by design by the Marlin team, but realistically that needs to be considered **after** the G29 data has been saved. If for instance I needed to change to a different nozzle that was not the exact same length as the current one, I need to be able to reset the nozzle-bed offset **and** have the G29-Mesh take that into consideration.

Is there anything I can do to make that happen?

Thanks
p.s. The 6mm-aluminum plates for the effector etc are machined on a commercial CNC-milling machine, so I know they are all identical when bolted to the effector.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@FlailAway please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek