MarlinFirmware/Marlin #9304: Processing + g code + marlin

donal1, 2018-01-23:
Is it possible to set up a com port to send g code using processing to connect to an arduino mega with ramps board running marlin? 

I just wanna be able to automatically send individual g codes like G1 Z10 from processing, eventually run processing from a pi. The machine I'm working on isn't a printer or a cnc so I don't want to have to use octoprint or grbl. 

(Remeber reading something similar in this forum but I can't seem to find it now)

p3p, 2018-01-24:
Processing can open and communicate over Serial ports, this is the only requirement for creating a control application, at a basic level the protocol is very simple just send the command "G1 Z10" followed by a newline, then wait for an "ok" response, If you need help with the implementation you should probably ask at a more appropriate place as you are more likely to get a response.

donal1, 2018-01-24:
Oh perfect, if you could recommend a forum that would be great. I'm very new to this. 

Thank you

revilor, 2018-01-26:
How about feeding the internet search engine of your choice with a query like "processing com port" or "processing com port arduino" to get you started?

closed 2018-02-14 by thinkyhead