MarlinFirmware/Marlin #9305: Z Axis Jerking down (Not from jerk setting) While probing M48/G29

Custom3DProducts, 2018-01-23:
Creality CR-10 With BLTouch Z axis Bug Report
- Description: Creality CR-10 With BLTouch. Z jerks down about 1/20th turn then smoothly rotates down to finish the probe. Once complete it smoothly rotates back up and repeats.
Youtube Video showing the jerking [https://youtu.be/ka9lZfgDax4](url)
Issue being Jerk seems to lessen as probes go on example #3 and on out of 10 (M48 P10) as the jerking lessens the z axis measurement becomes extremely accurate and repeatable but while it is jerking bad the measurement is all over.
- Expected behaviour: Z axis smoothly rotates down and back up in respect to jerk and acceleration settings. (Z Jerk 0.30 Z Acceleration 100)
- Actual behaviour: Z jerk down a small amount and then smoothly rotates down and back up.
- Steps to reproduce:
Use these configs and run M48 OR G29
[Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/1654031/Configuration_adv.h.zip)
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1654032/Configuration.h.zip)

Festivejelly, 2018-01-23:
I see this also. I can only reproduce this on my CR-10 and not on my Wanhao I3. Weird stuff.

Custom3DProducts, 2018-01-23:
-Youtube link fixed-

Festivejelly, 2018-01-24:
Right. This is what I think is happening... bare with me here.

User has a probing height set at 5mm
User homes in the middle of the bed. Printer homes and raises back to 5mm
User performs repeatability test for 10 iterations.

Iteration 1, lowers until probe is triggered. Probe triggers at 0.03

Printer now raises the Z axis by 5mm.

The problem now is we're now at 5.03 mm, and we know our Z axis can only move in multiples of 0.04. SO I think whats happening is that Marlin is first doing a small correction before moving down to probe.

Can this be confirmed? Im not sure where to look for it in the code. What I dont understand is why this is only happening on the CR-10. Maybe because the X axis gantry is heavier than on my other printers. So maybe its dropping slightly.

Custom3DProducts, 2018-01-24:
@Festivejelly I dont believe marlin adds a correction in before it decends. At least not like that jerk we are seeing. the jerk seems to be depended on something else because the measurements go from good to bad on the probe following the first jerk. not before as would be if it were jerking to correct height after a high or low probe measurement. 
Also the CR10 Z axis moves in multiples of 0.04mm for a full step and the cr10 has 16 micro steps so it really moves 0.0025 per step that also being the resolution of the z axis.

houseofbugs, 2018-02-05:
Did you rule out Z VREF being too low? You may not have enough torque to keep in a partial step.

thinkyhead, 2018-02-06:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch in case the issue is already fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

boelle, 2019-02-20:
@Custom3DProducts Please test with the latest bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek