MarlinFirmware/Marlin #9306: [TMC2130] Stealth Chop not so Stealthy

migeorge, 2018-01-23:
I have TMC2130 drivers setup on my X and Y axes with stealthChop enabled and I expected them to be much quieter.

I'm currently running the following configuration:
  - MKS Gen 1.4 @ 24V
  - Stepper Motors = NEMA 17 1.7A 1.8 deg
  - X and Y @ 1000mA current (/w active cooling and heatsinks)

I know there are some more advanced settings looking at the TMC2130 library by @teemuatlut but I'm unsure what I should be looking to tune for stealthChop sound level.

Here is a video that illustrates the sound I'm hearing:
https://photos.app.goo.gl/7cAH9gikHJ1eESQW2

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1654205/migeorge-configuration.zip)

oakdesign, 2018-01-23:
Genuine Watterott or Chinese TMC2130?

brisma, 2018-01-23:
Can you post the result of the M122 command? Only M122, without S0/S1.
Just to be sure that the SPI connection is working correctly :)

migeorge, 2018-01-23:
@oakdesign These are genuine Watterott.
@brisma I can't get the output right now but I can post it later, It does seem that the SPI connection is responding, I remember checking M122 and the currents are set correctly and it's reporting stealthChop as enabled.

At 1000mA my Y axis starts double stepping even though I'm well within the range of the motor, so I dropped X and Y to 900mA. They aren't double stepping there, but still noisy.

benny64, 2018-01-23:
I have the same problem, switching between stealthchop and spreadcycle no difference, watterott drivers

migeorge, 2018-01-24:
Here is the output I'm getting from M122:
```
READ: 		X	Y
READ: Enabled		false	false
READ: Set current	900	900
READ: RMS current	887	887
READ: MAX current	1251	1251
READ: Run current	28/31	28/31
READ: Hold current	8/31	8/31
READ: CS actual		8/31	8/31
READ: PWM scale	0	1
READ: vsense		1=.18	1=.18
READ: stealthChop	true	true
READ: msteps		16	16
READ: tstep		1048575	1048575
READ: pwm
READ: threshold		0	0
READ: [mm/s]		-	-
READ: OT prewarn	false	false
READ: OT prewarn has
READ: been triggered	false	false
READ: off time		5	5
READ: blank time	24	24
READ: hysterisis
READ: -end		2	2
READ: -start		3	3
READ: Stallguard thrs	2	3
READ: DRVSTATUS	X	Y
READ: stallguard			
READ: sg_result		0	0
READ: fsactive			
READ: stst		X	X
READ: olb			
READ: ola			
READ: s2gb			
READ: s2ga			
READ: otpw			
READ: ot			
READ: Driver registers:
READ: 	X = 0x80:08:00:00
READ: 	Y = 0x80:08:00:00
```

sh4jonny, 2018-01-24:
@migeorge try to increase this threshold
READ: threshold		0	0
to 150 mmps

sh4jonny, 2018-01-24:
@migeorge add to your config
#define HYBRID_THRESHOLD
#define X_HYBRID_THRESHOLD     150  // [mm/s]
#define Y_HYBRID_THRESHOLD     150

teemuatlut, 2018-01-24:
Mostly what I heard from your video were the fans and I'd say that your printing moves are pretty quiet already. If you mean the occasional high pitched "chirps", those are your travel moves that are nearing the maximum stepping frequency of stealthChop. The mode works best and is intended for slow to medium speed moves.

migeorge, 2018-01-24:
Okay, fair enough. This is the first printer I've put them on. 

I think my last issue is that I'm getting a lot of shifting on the Y axis but if I increase the current the axis starts double stepping and traveling twice as far as it should, has anyone seen that before?

migeorge, 2018-01-25:
Figured out I was experiencing a mechanical issue causing the Y axis shifting. All is well after adjusting the axis.

closed 2018-01-25 by migeorge

viktorglekler, 2019-08-11:
Hi all! I have the same problem with my tmc2208 and marlin 2.0. I switched from smoothieware to marlin and now my motors sound like on your video. On smoothieware they were really quiet! With the same current, acceleration and speed settings. What can be the problem? TMC library?