MarlinFirmware/Marlin #9308: Filament Runout Sensor not working on MKS Gen v1.4

BlocksTechnology, 2018-01-23:
I'm using an mks gen v1.4 to control my printer and i connected a micro-switch to the D4 pin on the board as seen on the picture.
[image]

And I change my configuration file to this.

`/**
 * Filament Runout Sensor
 * A mechanical or opto endstop is used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN.
 * For other boards you may need to define FIL_RUNOUT_PIN.
 * By default the firmware assumes HIGH = has filament, LOW = ran out
 */
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_PIN 4
  #define FIL_RUNOUT_INVERTING false // set to true to invert the logic of the sensor.
  #define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
`
I already tried to change the FIL_RUNOUT_INVERTING  to true and false but nothing seems to work.

Alissonverd, 2018-01-23:
Try uncomment the lines of RUNOUT and use the correct Pin in pins_RAMPS file.

BlocksTechnology, 2018-01-24:
Thanks for the answer Alissonverd but i have everything uncommented and I connected the filament runout sensor in the correct pin. and it still dont work

Alissonverd, 2018-01-26:
Test with older firmware version, if you have one.

hussainsail2002, 2018-11-28:
I have the exact same board, you don not need to use define Fil_runout_pin , it automatically takes the pin 4, make sure you connect the 3 pin connector to the D4 pin on the MKS board, the positive and negative pin need to be interchanged and you should be fine. Make sure that the advanced pause option is enabled.

I have done this on marlin 2.0 and it works fine. You may have to invert the logic.

Hope this helps

sl1pkn07, 2019-09-08:
i have the same issue with the Re-ARM+Ramps 1.4

connected to D4 (P1_18) (Servo 3) do nothing. set up to Z Max endtop pin (P1_28) the trigger works. but not when printer is running (print from SD) (never run the M600 code)