MarlinFirmware/Marlin #9309: [2.0.x] graphic lcd status screen

maukcc, 2018-01-23:
We are trying to get our lcd working
It is a U8GLIB_DOGM128_2X display on a Arduino due, on the same spi line as the sdcard
It shows the custom bootscreen, then the marlin bootscreen and then wants to go to the status screen.
But it seems it wants to put all data of the status screen into the status message on the bottom of the lcd.

AnHardt, 2018-01-23:
Did you try some of the u8glib examples on your hardware?

maukcc, 2018-01-23:
No, but MK4duo works with the lcd.
And basically Marlin is working as well, it just tries to cramp the whole display into the bottom 8(ish) pixels (status bar)

maukcc, 2018-01-23:
When SDSUPPORT is off, the display works

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-2.0.x`) branch to see if we got this working. If so then we can close this. Otherwise, more to do…

maukcc, 2018-02-20:
Sorry, more to do... :)
It is different, as it goes from the boot screen into the status screen now, and then tries to cram everything into the bottom line

thinkyhead, 2018-02-20:
Try unsetting `SD_DETECT_PIN` and see if it prevents the screen glitches. Also try the same with `LCD_SDSS` and `SDSS`. If any of these changes makes the screen work then it indicates a pin conflict.

Just add these lines to the bottom of your pins file:
```cpp
#undef SD_DETECT_PIN
#undef LCD_SDSS
#undef SDSS
```

Bob-the-Kuhn, 2018-02-20:
Exactly what LCD are you using?  Are you using the "RepRapDiscount FULL GRAPHIC Smart Controller"

maukcc, 2018-02-21:
@thinkyhead 
When I do #undef SDSS I get a lot of errors, the other 2 make no difference. The SDSS pin is 53 as it always has been. The SCK,MOSI and MISO have been diverted to the due SPI header.

@Bob-the-Kuhn 
The display is the CartesioUI, 
U8GLIB_DOGM128_2X u8g(DOGLCD_CS, DOGLCD_A0); // 4 stripes

thinkyhead, 2018-02-23:
Hmm… This is beyond my expertise to solve. Timing, voltage, wire length, RFI… So many potential issues.

Bob-the-Kuhn, 2018-03-08:
> on the same spi line as the sdcard

The current DUE code doesn't support this.  PR #9961 will resolve this.  

We're waiting on a tester to see if the USB mass storage is broken by the code changes.  Hopefully we'll know in a day or two.

If you want to test the new code then grab the ZIP file that's almost at the end of the PR.

Bob-the-Kuhn, 2018-03-11:
Please test the latest bugfix.  I think it fixes this issue.

maukcc, 2018-05-12:
The LCD works as expected. I can not read from the SD card, but that could be faulty wiring.

closed 2018-05-12 by maukcc