MarlinFirmware/Marlin #9313: Ramps+Mega or newer MKS boards better supported? Equally stable?

the1snm, 2018-01-23:
My stock control board from my Anet A8 blew the Y axis stepper driver recently and I need a new board. There are 1.2 bazilion duet Ramps/Mega combinations out there and a number of the MKS boards as well. I'm still fairly new to printing but the duet combo seems pretty standard out there and lower priced but then I stated seeing I'm gonna need to rework my wires for my steppers and everything else. This is where the MKS all in one board seemed to step in. I can use my old JST's to retain plug and play plus I get replaceable stepper drivers if it happens again. The question though; are the MKS boards supported or better stated equally stable under Marlin as the 2 part systems?

I've got an MKS in route but starting to second guess it wondering if I should have gone the more traditional route.

Bob-the-Kuhn, 2018-01-23:
The MKS boards are supported by Marlin.  Lots of people use them.

The only issue I've seen is the connector housing for the LCD display may be rotated 180 degrees from the LCD.

What LCD are you using now?

the1snm, 2018-01-23:
Thanks for the response, right now i just have the stock 5 button 2004 display. I'll upgrade to the rotary version or the 12864 in short time. I didn't want to introduce upgrades at the same time I'm repairing as I've learned that mistake before. For example I've got my E3D bowden ready to go with all the parts printed ready to install but now it waits. I was actually planning on installing the bowden setup the day my stepper died.

the1snm, 2018-01-24:
I didn't see a sample profile for the board I bought I assume I can use the same or very similar config as my Anet Marlin firmware. Should I be using a different starting point?

Bob-the-Kuhn, 2018-01-24:
Use your ANET configuration as the starting point.  

You may need to adjust the **steps_per_unit** if the microstep jumpers are set differently.

I doubt that the 5 buttom ANET LCD will be useable with the MKS board.  The ANET LCD uses an analog input for the five buttons.  I  doubt that the MKS board has anything similar.  Best if you order your 12864 right away.

the1snm, 2018-01-25:
Thanks Bob, after some tinkering last night I found the same thing as well regarding the LCD. I ordered a new 12864 and it's on its way. After multiple tries I was able to get a sketch to finally compile and load but it's far from working so I have lots of fixing to do. I could not get my previous Anet sketch to compile for this board as it would throw a multiple errors. I finally got a stock the default config to accept. The best I got was a couple steppers to respond to Octopi and wow do they sound different with this board and the DRV8825 steppers. I made sure to take the first 2 sets of pins off under each driver to drop it back to 1/16th steps as well. I noticed that after moving a stepper it emitted a low hum or whine unless I turned off motors in Octopi. I imagine that's a result of something I did wrong :) 

I'm surprised actually how different the code looks on the default ramps config.h vs the Anet version of the config.h file I was using. It's gonna take some time to parse through it I imagine. Are there any templates under examples that could start me closer? I imagined the Anet one would work after changing the board type but it just then complained about the LCD next.

the1snm, 2018-01-26:
OK I'm getting much closer now and have a pretty solid FW loaded. I'm able to control all the motors, extruder and bed heat up nicely and all the end stops work as expected. The only thing noggin scratching me now is the hookups for the fans. Based on some research I see where others have hooked them up to the 12V and 5V outputs on the bottom of the board near the USB. Problem is I don't see any reference to those pins in the Pins_RAMPS files so I don't know how to associate those with the fans. Nor do I know if they can handle PWM to control the speed of the fans. Anyone have thoughts?
[image]

Bob-the-Kuhn, 2018-01-26:
The D9 - pin controls the fan.  

If the fan uses the same power as the motors then connect the + lead of the fan to the D9 + pin or to the 12-24V pins you circled.

If the fan uses 5V power then the + wire of the fan goes to the +5V pins you circled.

D9  can be run by either a hardware or a software PWM.  In Configuration.h enable either **FAST_PWM_FAN** or **FAN_SOFT_PWM** as desired.

the1snm, 2018-01-26:
Thanks a bunch Bob, I woke in the middle of the night thinking duh the blower fan should go to D9 so the GCode can control it. Thanks for confirming that :) Regarding the extruder fan, the easy route is to plug it in to the 12V and leave it on all the time is there a way to use the #define E0_AUTO_FAN_PIN on this one  with that 12V plug or to re-purpose another one (I just don't see another 12V other than D7 for extruder 1)

closed 2018-01-26 by the1snm