MarlinFirmware/Marlin #9316: ### Bug Report - UBL Mesh issues with 1.1.8

modem7, 2018-01-23:
### Bug Report
- Description: Mesh is not using full bed size
- Expected behaviour: Mesh uses full bed size
- Actual behaviour: Mesh is not using full bed size
- Steps to reproduce:
  - Building mesh in UBL configuration

Updated firmware, cloned settings. 

Wanhao i3 v2.1, 200x200x180 w/ BLTouch.

If I'm being an idiot, please let me know!

Thanks!

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

-----------------------------------------------------------

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1657801/config.zip)

Festivejelly, 2018-01-23:
Does this same behaviour exist in an earlier version? Such as 1.1.5? I run this version with a 5mm inset and have no issues.

modem7, 2018-01-23:
The 10mm inset was just a test I did, but yes, previous version had zero issues

Here's my config from earlier version (prior to update) 

[config early.zip](https://github.com/MarlinFirmware/Marlin/files/1657843/config.early.zip)

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

Roxy-3D, 2018-02-14:
I believe this issue came about with the change to use BED_SIZE.    It was fixed roughly a month ago.

modem7, 2018-02-14:
Will give it a whirl this weekend to test!

thinkyhead, 2018-02-23:
We can re-open this if the problem recurs.

closed 2018-02-23 by thinkyhead