MarlinFirmware/Marlin #9318: Noisy motors with the TMC2208

liberssky, 2018-01-23:
- I have an AZSMZ MINI with 5x TMC2208 stepsticks on a corexy machine, recently I switched to Marlin 2.0.x and while printing I noticed the X/Y stepper are really noisy during fast moves, that is a high pitched stepper noise. I have the stepsticks in spreadcycle mode so I do not expect it to be completely silent, but I never had this issue before I switched from Smoothieware to Marlin. What could be the cause of that? Other than changing the stock steps per unit values and 'BLOCK_BUFFER_SIZE' from 16 to 32 I haven't done anything else that could be causing this. 

- Bonus question: Is Marlin 2.0.x with HAL running on a 32bit board capable of reaching higher printing speeds than just running Marlin on an 8bit board?

teemuatlut, 2018-01-24:
The spreadCycle parameters could use some work but I haven't had the time just yet and I've only had my scope for a few months.
Does smoothieware even support other TMC drivers than 26x or 2660?

liberssky, 2018-01-24:
I should clarify that I don't have the special functions for the TMC drivers turned on in Marlin, neither had I in smoothieware. I was just using them as any other stepsticks, at 16 microsteps (interpolated to 256) and on smoothie they were really quiet but on Marlin they are loud.

p3p, 2018-01-24:
I'm unsure why you would get more resonance at the same step frequency with Marlin than with Smoothie, (without a lot more details it's hard to even guess), my first thought would be perhaps you hit the double step frequency but that's unlikely? it's 80kHz with an LPC1768.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-2.0.x` branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

liberssky, 2018-02-15:
I tried a different config, different acceleration values and the problem went away, closing.

closed 2018-02-15 by liberssky

maar1201gh, 2018-12-13:
@ghost, i use TMC  as any other stepsticks as same as You on my corexy and upgrading to last Marlin bugfix-2.0.x i have much noise on X,Y stepper.
Could you inform what you did to resolve you problem? Thanks

viktorglekler, 2019-08-08:
Hi! I have the same problem! Today I switched from smoothieware to marlin and noticed that my motors sound loud. I have tried different settings in Configuration.h, but nothing help. What exactly did you corrected in Marlin?