MarlinFirmware/Marlin #9320: Bltouch doesn't stop the movement .

giachy96, 2018-01-24:
Hi everyone. I have installed my bltouch since a month ago and it was working very good but since yesterday it doesn't stop the movement of the z axis.
It detect the plate because it retract the pin but don't stop. I tried with many configuration and command like the M280 series but nothing change.
Obviously my M119 command say it is Open when the pin is up and I can't read with this command if it triggered ( if I understand correctly what I have read here).
What can I do?

Bob-the-Kuhn, 2018-01-24:
Correct - M119 will say open unless you're really lucky in your timing or you're in the test mode.

The BLTouch automatically retracts the pin when it is triggered so the probe knows that something happened.  

There's a servo probe subtest in the M43 command.  In **Configuration_adv.h** enable **PINS_DEBUGGING** and then issue **M43 S**.  After that follow the prompts on the host screen.

giachy96, 2018-01-24:
this is what i have received for answer form the M43 command
READ: PIN:   0   Port: E0        RXD                         protected 
READ: PIN:   1   Port: E1        TXD                         protected 
READ: PIN:   2   Port: E4        X_MAX_PIN                   protected 
READ: PIN:   3   Port: E5        X_MIN_PIN                   protected 
READ: PIN:   4   Port: G5        FIL_RUNOUT_PIN              Input  = 1    TIMER0B   PWM:   128    WGM: 3    COM0B: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled  
READ: .                          SERVO3_PIN                  Input  = 1
READ: PIN:   5   Port: E3        SERVO2_PIN                  Input  = 1    TIMER3A   PWM:  -25536    WGM: 0    COM3A: 0    CS: 2    TCCR3A: 0    TCCR3B: 2    TIMSK3: 2   non-standard PWM mode   compare interrupt enabled  
READ: PIN:   6   Port: H3        SERVO1_PIN                  Input  = 1    TIMER4A   PWM:     0    WGM: 1    COM4A: 0    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0  
READ: PIN:   7   Port: H4        FAN1_PIN                    protected 
READ: .                          MOSFET_D_PIN                protected 
READ: PIN:   8   Port: H5        HEATER_BED_PIN              protected 
READ: .                          RAMPS_D8_PIN                protected 
READ: PIN:   9   Port: H6        FAN_PIN                     protected 
READ: .                          RAMPS_D9_PIN                protected 
READ: PIN:  10   Port: B4        HEATER_0_PIN                protected 
READ: .                          RAMPS_D10_PIN               protected 
READ: PIN:  11   Port: B5        SERVO0_PIN                  Output = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled  
READ: PIN:  12   Port: B6        PS_ON_PIN                   protected 
READ: PIN:  13   Port: B7        LED_PIN                     Input  = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 0    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled  
READ:  .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
READ: PIN:  14   Port: J1        Y_MIN_PIN                   protected 
READ: PIN:  15   Port: J0        Y_MAX_PIN                   protected 
READ: PIN:  16   Port: H1        LCD_PINS_RS                 Output = 1  
READ: PIN:  17   Port: H0        LCD_PINS_ENABLE             Output = 0  
READ: PIN:  18   Port: D3        Z_MIN_PIN                   protected 
READ: PIN:  19   Port: D2        Z_MAX_PIN                   protected 
READ: PIN:  20   Port: D1        SDA                         Input  = 1  
READ: PIN:  21   Port: D0        SCL                         Input  = 1  
READ: PIN:  22   Port: A0        <unused/unknown>            Input  = 1  
READ: PIN:  23   Port: A1        LCD_PINS_D4                 Output = 0  
READ: PIN:  24   Port: A2        E0_ENABLE_PIN               protected 
READ: PIN:  25   Port: A3        LCD_PINS_D5                 Output = 0  
READ: PIN:  26   Port: A4        E0_STEP_PIN                 protected 
READ: PIN:  27   Port: A5        LCD_PINS_D6                 Output = 1  
READ: PIN:  28   Port: A6        E0_DIR_PIN                  protected 
READ: PIN:  29   Port: A7        LCD_PINS_D7                 Output = 0  
READ: PIN:  30   Port: C7        E1_ENABLE_PIN               Output = 1  
READ: PIN:  31   Port: C6        BTN_EN1                     Input  = 1  
READ: PIN:  32   Port: C5        <unused/unknown>            Input  = 1  
READ: PIN:  33   Port: C4        BTN_EN2                     Input  = 1  
READ: PIN:  34   Port: C3        E1_DIR_PIN                  Output = 0  
READ: PIN:  35   Port: C2        BTN_ENC                     Input  = 1  
READ: PIN:  36   Port: C1        E1_STEP_PIN                 Output = 0  
READ: PIN:  37   Port: C0        BEEPER_PIN                  Output = 0  
READ: PIN:  38   Port: D7        X_ENABLE_PIN                protected 
READ: PIN:  39   Port: G2        <unused/unknown>            Input  = 1  
READ: PIN:  40   Port: G1        Z_CS_PIN                    Input  = 1  
READ: PIN:  41   Port: G0        KILL_PIN                    Input  = 1  
READ: PIN:  42   Port: L7        E0_CS_PIN                   Input  = 0  
READ: PIN:  43   Port: L6        <unused/unknown>            Input  = 0  
READ: PIN:  44   Port: L5        E1_CS_PIN                   Input  = 0    TIMER5C   PWM:     0    WGM: 1    COM5C: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0  
READ: PIN:  45   Port: L4        <unused/unknown>            Input  = 0    TIMER5B   PWM:     0    WGM: 1    COM5B: 0    CS: 3    TCCR5A: 1    TCCR5B: 3    TIMSK5: 0  
READ: PIN:  46   Port: L3        Z_STEP_PIN                  protected 
READ: PIN:  47   Port: L2        <unused/unknown>            Input  = 0  
READ: PIN:  48   Port: L1        Z_DIR_PIN                   protected 
READ: PIN:  49   Port: L0        SD_DETECT_PIN               Input  = 1  
READ: .                          Y_CS_PIN                    Input  = 1
READ: PIN:  50   Port: B3        AVR_MISO_PIN                Input  = 0  
READ: .                          MISO_PIN                    Input  = 0
READ: PIN:  51   Port: B2        AVR_MOSI_PIN                Output = 1  
READ: .                          MOSI_PIN                    Output = 1
[... cut ...]
READ: PIN:  63   Port: K1 (A 9)  <unused/unknown>   Analog in =   796   Input  = 1  
READ: PIN:  64   Port: K2 (A10)  <unused/unknown>   Analog in =   929   Input  = 1  
READ: PIN:  65   Port: K3 (A11)  <unused/unknown>   Analog in =   760   Input  = 1  
READ: PIN:  66   Port: K4 (A12)  MAX6675_SS                  Input  = 1  
READ: PIN:  67   Port: K5 (A13)  TEMP_0_PIN                  protected 
READ: PIN:  68   Port: K6 (A14)  TEMP_BED_PIN                protected 
READ: PIN:  69   Port: K7 (A15)  TEMP_1_PIN                  Analog in =  1023

giachy96, 2018-01-24:
here my configuration files. in a zip folder
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1658795/configuration.zip)

giachy96, 2018-01-24:
I am getting mad ! Nothing is working with this probe

Bob-the-Kuhn, 2018-01-24:
You issued "M43".  Try "M43 S".

----

Maybe there's a cable problem.  

M119 should report triggered if you remove the cable at the BLTouch end.  

If it says open then there's a problem.   Remove the cable at the RAMPS end.  If M119 reports triggered then the cable is bad (most likely signal is shorted to ground).  If M119 reports open then either the RAMPS/2560 is bad or you're not running the config you posted.

----

If the above checks out OK then try this:
```
	Testing a BLTouch with serial number:
	1. Connect BLTouch, power up controller and upload the firmware
	2. Blue LED will be ON if a servo signal is present.  Blue LED is faint.
	3. Issue "M119" ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used. Also check that Z_MIN_ENDSTOP_INVERTING and Z_MIN_PROBE_ENDSTOP_INVERTING are both set to "false".
	4. Issue "M280 P0 S10" ; should deploy the probe and the orange LED will be OFF.
	5. Issue "M119" ; should return OPEN
	6. Issue "M280 P0 S90" ; should stow the probe and the orange LED will be ON
	7. Issue "M280 P0 S60" ; puts it into the M119 test mode.
	8. The probe should remain stowed and the blue LED should be OFF. If it's ON then the unit needs to be adjusted.
	9. Issue "M119" ; should return TRIGGERED
	10. Issue "M280 P0 S160" ; returns it to normal operation
	11. Issue "M119" ; should return OPEN.

```

GMagician, 2018-01-24:
@Bob-the-Kuhn in configuration ENDSTOPPULLUPS is disabled, maybe this can have something to do with the issue?

Bob-the-Kuhn, 2018-01-24:
I doubt it but enabling it won't hurt.

The ENDSTOPPULLUP is an internal 50k (approximately) device within the CPU.  Usually there's a 4.7k pullup on the board because the 50K tends to allow noise.  MKS doesn't publish their schematics so I can't verify that there is a 4.7K on their board.

giachy96, 2018-01-25:
Thanks for all the support. I tried to remove the cable from the bltouch end or form the  board end and the m119 report triggered correctly

giachy96, 2018-01-25:
I tried the script you posted here. All seems ok until the command M280 P0 S60. With this nothing happened:/

giachy96, 2018-01-25:
I tried to set in the pinRamps.h the #define  Z_MIN_PROBE_PIN  Z_MIN_PIN

giachy96, 2018-01-25:
Here the result of the M43 S command 
SENT: M43 S
READ: Servo probe test
READ: .  using index:  0
READ: .  deploy angle: 10
READ: .  stow angle:   90
READ: . probe uses Z_MIN pin: 18
READ: . uses Z_MIN_ENDSTOP_INVERTING (ignores Z_MIN_PROBE_ENDSTOP_INVERTING)
READ: . Z_MIN_ENDSTOP_INVERTING: false
READ: . deploy & stow 4 times
SENT: M105
SENT: M105
READ: please trigger probe
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
READ: trigger not detected

it dosen't detect when i try to trigger the pin manually in this test.
@Bob-the-Kuhn

Bob-the-Kuhn, 2018-01-25:
If the probe deployed & retracted during the M43 S test then the servo signal is getting to the BLTouch properly.  Since there is no trigger then the BLTouch is bad.

If the probe didn't deploy & retract then either the servo signal isn't getting to the BLTouch or the BLTouch is bad.  Check your servo wiring.

giachy96, 2018-01-28:
yesterday a new BLtouch arrived at my home. Today i have install it and try it. Bad News.  it dosen't work. 
i don't know what can i do. Now the Bltouch must work i think because it's new. It's a firmware or motherboard problem? @Bob-the-Kuhn

giachy96, 2018-01-28:
Again the command M280 P0 S60 dosen't work.

GMagician, 2018-01-28:
Have you tried to enable ENDSTOP_INTERRUPTS_FEATURE (if your board support it) ?
Maybe 5ms signal from BLtouch is not enough to detect trigger.
Since endstop are in polling and maybe firmware has become slower (if you upgraded)

giachy96, 2018-01-28:
I Will try it immediatly !

giachy96, 2018-01-28:
You are my saviour ! Now it is working! There are others problems but I will fix them plasurly:)

GMagician, 2018-01-28:
@Bob-the-Kuhn what do you think about adding sanity check to prevent bl-touch without interrupt?
Since 5ms seems to few to let polling detect signal change.
I don't have enough vision on marlin history to decide it.

Bob-the-Kuhn, 2018-01-28:
I'm really glad to hear that the BLTouch is working properly.

I'm struggling to understand how enabling ENDSTOP_INTERRUPTS_FEATURE would fix the problem.

If the other system were broke, broke, broke then we'd be flooded with problems.  Something must be marginal.

I'd dearly love to put an oscilloscope on the system to see what's happening.

The interrupt system requires the pulse to be active for less time than the other.

Maybe there's a power problem.  The BLTouch requires a lot more current when deploying and retracting the probe.  Try running separate power & ground wires to the BLTouch.  That's helped others.

Double check the 5V/3.3V jumper on the BLTouch.  It should be intact since the 2560 is a 5V system.

giachy96, 2018-01-28:
I have a little oscilloscope if you want I will check what you will ask me. yes the jumpers is on 5v.
I am thinking that the other one (that I have resend to the seller) was good like the new one.

Bob-the-Kuhn, 2018-01-28:
Put the 'scope on the BLTouch's output and see what the pulse looks like.  It should be a 5 or 10mS pulse that's above 3V.

closed 2018-02-14 by thinkyhead

Amaynard89, 2019-02-09:
shout out to thinkyhead, bob the Kuhn, researched the shit out of your comment and responses. because of you guys my 3dtouch is working great!