MarlinFirmware/Marlin #9321: Babystepping immediately undone

DonStauffer, 2018-01-24:
- Description: With bilinear bed leveling under Marlin 1.1.8, babystepping to change the nozzle vertical position will go back to where it was within a few seconds. This happens repeatedly but eventually stops happening, so you can get the desired height if you keep trying. But you have to keep cranking the knob and end up with an outrageously large babystepping amount, like 2 to 3 mm.
- Expected behaviour: Babystepping moves the nozzle to a new height and it stays there for the remainder of the print.
- Actual behaviour: Half the time when you move the nozzle to a new height, a few seconds later it goes back to its old height.
- Steps to reproduce:
  - Start a print
  - After the G29 finishes executing, press the controller knob twice to enter babystepping
  - When plastic starts extruding, turn the knob to change the Z height
  - Watch the Z height go back to what it was a second later

Using EZABL by TH3D, so the configuration files are customized.
[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/1658093/MarlinConfig.zip)

thinkyhead, 2018-02-14:
Known issue.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@DonStauffer Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek