MarlinFirmware/Marlin #9322: Nozzle too high an unpredictable amount

DonStauffer, 2018-01-24:
- Description: Using ABL under Marlin 1.1.8, after the G29 the nozzle starts too high even if the Z offset was set correctly. Monitoring babystepping and adjusting Z offset for the next print results in unpredictable nozzle height, but typically still too high by somewhat less. 
- Expected behaviour: When Z offset is set correctly as a negative value whose absolute value is the distance from nozzle to print surface when the probe triggers, babystepping gets used just to compress the first layer an appropriate amount for the material being printed. This is usually a fraction of the layer height. Adjusting the Z offset by this amount should make it so the next print requires little or no babystepping to achieve the same compression.
- Actual behaviour: When the Z offset is set correctly, the nozzle is so high the filament can be seen dropping from nozzle to bed. Many full turns of the knob are necessary to babystep down to a reasonable height. Reading the babystepping value and adjusting Z offset that much before the next print does not ensure the next print starts anywhere close to the right height.
- Steps to reproduce:
  - Set the Z offset according to the procedure given in Marlin's M851 command documentation
  - Start a print
  - Wait for the G29 to finish
  - Watch where the nozzle is when the plastic starts flowing
  - Click the controller twice to get into babystepping, and start cranking the knob until the nozzle gets to the right height
  - Make a note of the babystepping value
  - After the print finishes, add the babystepping value to the Z offset and set that as the new value using M851
  - Start another print and watch the same process repeat, probably with the nozzle starting out a little lower, but not close to low enough

[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/1658115/MarlinConfig.zip)

Festivejelly, 2018-01-24:
After setting the offset are you doing M500 to save it to eeprom?

DonStauffer, 2018-01-24:
Yes, I'm saving the offset.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@DonStauffer Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek