MarlinFirmware/Marlin #9324: Linear advance doesn't work with some bowdens, is some lower value better than disabled?

Itox001, 2018-01-24:
Calling @Sebastianv650 here. Not really an issue, rather some advice on Linear Advance.

 Even though I have a relatively short, tight tolerance Capricorn bowden tube (around 350 mm) on a Titan clone and am running 2.8 mm retraction without stringing, the linear advance test pattern doesn't look good until around K=265.

Problem is, when I try to print anything with that K value, the extruder compensates so much that it ends up not extruding anything and making ugly noises. So my question is, if I use a k number lower than what the test pattern suggest, will I still see improvements in prints, just not as perfect? Or would it be better to just leave it disabled? Currently running latest Marlin stable release, 1.1.8.

Thanks and feel free to close the issue when @Sebastianv650 responds.

Sebastianv650, 2018-01-24:
Yes you should get better results with a K value where the extruder isn't loosing steps compared to K=0 (no advance).
The next version of LIN_ADVANCE will compare the executed speed change to the extruder jerk at block junctions, so faulty prints should be avoided and the user gets a serial message with a note that K is set too high.

Itox001, 2018-01-24:
Thank you Sebastian. This lin_advance 2 will have better compatibility with bowden setups or will it just warn you that it doesn't work? Cheers.

Sebastianv650, 2018-01-24:
It will give a warning when the needed stepper speed change is above the given extruder jerk limit. There is not much I can do more for bowden, each K will result in some need speed change for pressure adaption. I can't change reality, I have tried it :-)

I'm open for ideas in #9048, but it's a complicated topic. I guess that's why even after a lot of years in 3D printing, there is no generall widely used pressure adaption available. Some experimental implementations and theories here and there, but that's all.

Itox001, 2018-01-24:
Just left an idea there, a simplification of pressure advance that perhaps could give good results in bowden systems, check it out. I'm closing the issue here 'cause you already answered my concerns. Cheers!

closed 2018-01-24 by Itox001