MarlinFirmware/Marlin #9328: Bltouch doesn't stop the movement .

giachy96, 2018-01-24:
Hi everyone. I have installed my bltouch since a month ago and it was working very good but since yesterday it doesn't stop the movement of the z axis.
It detect the plate because it retract the pin but don't stop. I tried with many configuration and command like the M280 series but nothing change.
Obviously my M119 command say it is Open when the pin is up and I can't read with this command if it triggered ( if I understand correctly what I have read here).
What can I do?

GMagician, 2018-01-24:
First you can try to connect a endstop to Z input and check if problem is board input

GMagician, 2018-01-24:
second: close 2 of the 3 issues

giachy96, 2018-01-24:
Sorry for the three issue . I will close 2 of them. I will try to check the motherboard with an endstop

closed 2018-01-24 by giachy96