MarlinFirmware/Marlin #9330: [Support Q] TMC2130 SENSORLESS_HOMING with deltabots?

Hywelmartin, 2018-01-24:
Is it possible to use SENSORLESS_HOMING with deltas?

 in * Configuration_adv.h it looks like this under // @section TMC2130, TMC2208

 //#define SENSORLESS_HOMING // TMC2130 only

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  8
    #define Y_HOMING_SENSITIVITY  8
  #endif

should it look like this??

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  8
    #define Y_HOMING_SENSITIVITY  8
    //#define Z_HOMING_SENSITIVITY  8 // WARNING Use only with deltas/ bots that home to max
  #endif

teemuatlut, 2018-01-24:
I don't have any experience with deltas nor do I have a machine to test on but you could use the TMC_ADV macro to set up additional parameters for the drivers.

Hywelmartin, 2018-01-24:
Well this isn't an additional parameter, it's already there. The only thing is that we can setup this function for X-stepper and Y-stepper but not for Z-stepper in the firmware.
Or am I wrong?
I currently don't own any 2130, they are in the mail...

teemuatlut, 2018-01-24:
You can setup the drivers any way you like through the TMC_ADV.
Like if you wanted to enable stealthChop for X and Y but not for Z and E. Then you'd put this in the TMC_ADV
```cpp
stepperZ.stealthChop(false); \
stepperE0.stealthChop(false); \
```
Or increase chopping frequency in spreadCycle:
```cpp
stepperX.off_time(4);\
```
The way sensorless homing works is Marlin will configure the threshold values on startup and set the stepping speeds so that stallGuard should be enabled when in spreadCycle mode.
There's an additional step when you have stealthChop enabled so that when homing, Marlin will switch to spreadCycle first, and then restore it after. This is because sensorless homing (stallGuard) works only when in spreadCycle mode.

I don't see why sensorless homing wouldn't work for deltas, but as I don't have any means of testing and developing for that frame type, I will not be putting effort into it at this time. Anyone is welcome to try it out though and I can give some direction on what should be changed to get it working on an additional stepper.

Hywelmartin, 2018-01-24:
It was my thinking when the option isn't present i Configuration_adv.h that it wasn't supported on all axis.....

by the way, where do I find TMC_ADV or is it tmc_macros.h??

Hywelmartin, 2018-01-24:
Could it be a "hidden" feature. I think most people that set up deltas starts with a configuration.h and a Configuration_adv.h from the folder example_configurations ..

teemuatlut, 2018-01-24:
That could be a possible way to set it up. I went through the config files looking for delta definitions and noticed that they're there only on the example file(s). Or just add a `#if ENABLED(DELTA)` to the Z stepper sensitivity.

mylife4aiurr, 2018-01-25:
Sensorless homing to home position. I'd try it. Could remove the endstops at top of printer, if it worked reliably. 

But im not brave enough to do sensorless homing for moves toward the hot bed. All 3 motors moving hotend to hotbed, once hot end hits the bed -  tmc's sense a skip and reverse?? Hotend wouldnt last long at all.

Hywelmartin, 2018-01-25:
And it could even be added to the sanity check of Configuration.h

 if  #define Z_HOME_DIR not = 1

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

Hywelmartin, 2018-01-31:
So now I have installed the tmc2130. and they work...

I can test sensor less homing on my delta aswell