MarlinFirmware/Marlin #9333: Advanced break

boubouexpress, 2018-01-24:
add feature break advanced cut off the heater the bed and hotend for a recovery after several hours of break

boubouexpress, 2018-01-24:
as the original firmware of the creality 3D CR 10 S

MasterPIC, 2018-01-24:
I'd suggest to add an hotend standby temperature, useful to change filament at optimal temperature instead of forcing the user to remove the filament at extruding temperature...
Then, after a firmware defined timeout, it could automatically cut off the heater.

Grogyan, 2018-01-25:
Cold pull temperatures are dependent on the material used.
I can see how this could be useful

MasterPIC, 2018-01-25:
Maybe opting for setting how many degrees to lower the temperature is a good workaround...

boubouexpress, 2018-01-29:
to print a large room, pausing during the night for reasons of security and noise and resume the next morning
the pause mode should have a break of the heater

MasterPIC, 2018-01-30:
@boubouexpress I agree with you about deactivating the heater after a timeout.
But I was suggesting to add also an intermediate temperature state (by lowering the temperature by x°C with x user defined) to properly change the filament...

thinkyhead, 2018-02-14:
@tcm0116 I suggest we get down with the heater timeout code and generalize it so it can be used in all pause-like procedures. Currently it only exists within `wait_for_filament_reload` but it appears completely ready to be deployed.

@boubouexpress Shame on you teasing us with the Creality firmware and then not providing a copy so we can study their implementation. We demand a tribute!

boubouexpress, 2018-02-15:
I'm not the source code just the compiled 

[CR-10S-500Â¤Áþð°┤‗-┤┤¤ÙËó╬─.zip](https://github.com/MarlinFirmware/Marlin/files/1726816/CR-10S-500A.Athd.-.UEo.zip)

closed 2018-02-15 by boubouexpress