MarlinFirmware/Marlin #9336: Head raises by Z_CLEARANCE_BETWEEN_PROBES after homing

Keavon, 2018-01-24:
I have `Z_CLEARANCE_BETWEEN_PROBES` set to 1mm so that my Z-probe raises itself 1mm between each probing in the auto-leveling procedure. Also after homing the Z axis, it checks its height once quickly, bumps, raises itself slightly, then probes again slowly. But when it finds the correct Z home height on `G28`, it then backs off by 1mm and ends the `G28` operation a distance of `Z_CLEARANCE_BETWEEN_PROBES` above the bed. The LCD display shows that Z is at 1, and I am able to manually move the Z axis back down to 0. This behavior is particularly annoying, as when I home the axes, I wish for it to actually remain homed. When the nozzle is elevated, filament begins oozing out of the nozzle and forming a bead. When the nozzle is cold, this bead has hardened and caused the head to crash 1mm before reaching the bed. How can I disable the behavior where the Z axis raises itself after homing?

AnHardt, 2018-01-24:
You can't!
Homing is not a 'go to' but a 'find where you are'.
The end position of Z is 0 + max(Z_CLEARANCE_BETWEEN_PROBES, Z_CLEARANCE_DEPLOY_PROBE)
Use G0 or G1 if you want to go somewhere.

Keavon, 2018-01-24:
If that's not part of the settings, could you direct me to a file and line number in the project source code to disable the post-homing raising behaviour?

revilor, 2018-01-24:
Depending on what code base you want to use Marlin_main.cpp with function gcode_G28 (1.1.*) or G28.cpp (2.0) would be the location to go.

But just for curiosity: why is it a problem to send G0 Z0 after G28 to move the hotend to the position you want it to be?

Keavon, 2018-01-29:
I am using Marlin `1.1.8`. I could not seem to find which line to modify because it appears the `gcode_G28` routine is far more complicated than I anticipated. I tried following all the functions it calls and checking which functions in the entire file reference the `Z_CLEARANCE_BETWEEN_PROBES` #define and none seem to do it. I'd much appreciate help tracking down what to modify (or making a feature request to add this to the configuration).

I can definitely add a `G0 Z0` command to my slicer's frontmatter but I also use Pronterface to move the axes around to get the printer all prepared prior to a print. When I hit Home, I have to also hit Z-1mm which is something I can easily forget. Filament will ooze out of the nozzle until I lower it to touch the bed. If I neglect to lower the Z axis back to 0 after homing, then disconnect from the printer or disable the nozzle heater, a 1mm tall globule of hardened plastic will be attached to the nozzle. This effectively lowers the nozzle clearance by 1mm, which will cause the head to crash into the print surface next time I home. Because the probe is connected to the extruder carriage, but the crashing printhead + hardened plastic is pushing the bed down, the probe will never trigger and the Z axis will continue attempting to lower. This happened twice.

Just intuitively, it seems like a command to home the axes should leave the axes in a homed position. It shouldn't decide to raise one axis by an arbitrary amount after the user asks to home the axis. It's also far less capable of causing damage or leaving the nozzle in a placement where it can drip plastic when it stays at its home position.

AnHardt, 2018-01-30:
Discover the power of (pronterface / RH / Octoprtint)  macros. Build your own 'Home'-button.

Try it for a while.

Delete it, because you will always scratch the bed when you forget to lift the nozzle before moving in X or Y.

thinkyhead, 2018-02-14:
Please attach your Configuration files. ZIP them first.

Keavon, 2018-02-17:
[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/1733513/Marlin_Config.zip)

thinkyhead, 2018-02-19:
@Keavon — The final raise is due to this section in `homeaxis`:
```cpp
  // Put away the Z probe
  #if HOMING_Z_WITH_PROBE
    if (axis == Z_AXIS && STOW_PROBE()) return;
  #endif
```
Marlin raises the probe based on settings before deploying or stowing. If you only want the probe to raise for deploy and not for stow, you can make this change to the `set_probe_deployed` function:
```diff
    // Make room for probe
-   do_probe_raise(_Z_CLEARANCE_DEPLOY_PROBE);
+   if (deploy) do_probe_raise(_Z_CLEARANCE_DEPLOY_PROBE);
```

closed 2018-02-19 by thinkyhead