MarlinFirmware/Marlin #9337: [FR] babystepping without an encoder?

filipgoc, 2018-01-24:
### Feature Request: Let babystepping be triggered without a rotary encoder. 

More and more printers and setups are moving into all sorts of touch screens, octoprints, astroprints, etc. 

The amazingly awesome babystepping (Z babystepping really), which has become a cornerstone of my ABL/UBL tweaking is not compatible with those 😢. 

I have seen this in a few posts before (how to assign buttons to babystepping etc...), but it always died off. 

I'm curious if you guys agree this would be a prudent future proofing move.

revilor, 2018-01-26:
Is M290 not working in the scenario you have in mind?

fiveangle, 2018-01-26:
> The amazingly awesome babystepping... is not compatible with those 😢.

Yeah, it certainly should be "compatible".  If you're using Octoprint for example, just install _Custom Control Editor_ plugin and put a couple buttons in OP like "**_Raise Nozzle 0.01_**", "**_Lower Nozzle 0.01_**" that execute `M290 Z0.01` and `M290 Z-0.01` respectively.  Done & Done.

studiodyne, 2018-02-11:
0.1 is enough 👍

filipgoc, 2018-02-11:
Oh, my bad. My post was from the time BEFORE I discovered M290, but I forgot about this thread. 

So... for all you people out there, use M290 cause it works great!

filipgoc, 2018-02-11:
> 0.1 is enough

Depends on the layer height doesn't it ;-)

closed 2018-02-11 by filipgoc

studiodyne, 2018-02-11:
0.1 steps is enough whatever the height except if there zre some mad guys that print with 0.05 mm lolll
1200 hours for a 1cm3 object loll