MarlinFirmware/Marlin #9338: Successful UBL leveling,  but while printing head crashes into bed

pmbroth, 2018-01-24:
Hello I have a weird situation.  I’m using mks gen 1.4 on a formbot trex plus printer.  Since approximately mid November releases,including 1.18   Build I have this issue.  I use UBL auto level. And the mesh builds with no problems ,  then when I print the head crashes into the bed after a minute of printing. The head crash occurs approximately at the same spot every time.   Any ideas what I need to do?  I’m using an inductance probe, which i fine tuned last night. Thank you in advance Brian

Roxy-3D, 2018-01-26:
Is the nozzle leaving the mesh?   Can you isolate the line of gcode that makes things go sour?

pmbroth, 2018-01-27:
Hello,   As it is printing, the nozzle is moving above the bed at a different times , and then it will crash down into the bed.  I have tried with several test stl, and the action continues as it prints the skirt, and crashes at different points while printing, it never gets though a 1 layer of skirt.  Is there anything I can I provide?

fr33dimension, 2018-01-31:
I experience a similar problem, after using UBL for months on ver 1.1.4 without a glitch, I flashed 1.1.8 yesterday. After reloading the mesh, which I saved as a text doc, it will not act as expected. My machine will go to the waiting positin G1 X190.4 Y15 Z10 F9000.0, will extrude 10mm, then lower and make the prime move G1 X190.400 Y0.000 Z0.27 F9000.000. But it does not go down towards the bed, it raises up and I have to emergency stop it, since it will just keep going. Now if I do these two commands manually, by entering them in the g code command line, the nozzle crashes into the bed. Also if UPL is deactivated, the machine moves/ acts normal. 
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1681641/Configuration.h.txt)

fr33dimension, 2018-01-31:
I think I figured this out but I am not sure what to change. The machine thinks the bed ends at the last probing position. My probe is offset 43mm to the left of the nozzle, my bed is 194mm wide, the last possible probing point is at about 151mm. That is where the machine goes nuts when printing.

pmbroth, 2018-01-31:
> <author@noreply.github.com>
> printing head crashes into bed (#9338)

Hello,

I do not see your comment on the board.

My bed is 400*609mm, you haver to change

  #define MESH_INSET 50

At l;east that is what I believe   Mine is 50 because of my clips.

fr33dimension, 2018-01-31:
I started a new issue "UBL offset", but wouldn't that move my entire mesh 50mm away from the start of the bed ? I didn't do that in earlier marlin version.

pmbroth, 2018-02-01:
I had to ensure that the head would not crash into. The clips holding my
prints to the glass.  So I had to create a perimeter offset of 50mm. Your I
set should be whatever you need it to be. I that does not help with your
issue then there is a problem. I ended up going back to billenial
leveling.  The ubl keeps causing the head to crash into the bed at random
places while printing.   I have tried diffferent all files just in case
something was funky.

Let me know how it goes

Regards

Brian

fr33dimension, 2018-02-03:
Thanks Brian,
I use the UBL since months and it works great. Something changed in 1.1.8 marlin that causes this issue.
Appreciate your reply.

Fri

thinkyhead, 2018-02-17:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

pmbroth, 2018-02-17:
Hello I pulled a code bug fix 1.1x last Saturday.  Still did the same thing.  Did anything change since last week?  I switched to bilinear leveling and all works perfectly.

Roxy-3D, 2018-02-17:
> Hello I pulled a code bug fix 1.1x last Saturday. Still did the same thing. Did anything change since last week? I switched to bilinear leveling and all works perfectly.

Yes.   SEGMENTED_MOVES was turned on by default.   That was cutting out UBL's lower level sanity checks on the generated Z-Correction values.   SEGMENTED_MOVES no longer affects UBL Cartesian machines.   (We should get the two features compatable with each other...  But right now...  SEGMENTED_MOVES is ignored even if specified.)

It is very possible (likely) the problem is gone now for you...

pmbroth, 2018-02-17:
I will try it today and let you know

pmbroth, 2018-02-17:
I just downloaded the latest Bugfix, and the UBL is now working perfectly!!    :-)  The head does not crash anymore into the bed...

closed 2018-02-24 by pmbroth