MarlinFirmware/Marlin #9339: ANet A8, BLTouch, Not Seeing Height Compensation after Leveling

SirEggbert, 2018-01-24:
------------------------------------------------------------
### Bug Report
- Description: After warm-up and leveling my initial layers are not level.
- Expected behavior: I would expect that after leveling and starting the print the initial layer would be level.
- Actual behavior: While I see the touch point print out of the leveling action, the printer does not seem to be using this information during the print.
```
Send: G29
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 -3.992 -3.740 -3.285
Recv:  1 -4.270 -3.982 -3.525
Recv:  2 -4.313 -4.183 -3.835
Recv: 
Recv: X:220.00 Y:195.00 Z:19.47 E:0.00 Count X:22000 Y:19500 Z:6320
Recv: ok
```

- Steps to reproduce:
  - Startup GCODE:
```
M280 P0 S160 ; BLTouch alarm release
M117 Auto Leveling ...
G4 P100 ; delay for BLTouch
G28; Home Axis
G29; Z-Probe the bed
M117 Priming Extruder ...
G0 X1 Y1 F3000 ; Go to front
G0 Z0.15 ; Drop to bed
G92 E0 ; zero the extruded length
G1 X100 E40 F500 ; Extrude 25mm of filament in a 4cm line
G92 E0 ; zero the extruded length
G1 E-5 F500 ; Retract a little
G1 X180 F500 ; Quickly wipe away from the filament line
M117 Printing ...
G1 Z5 ; Raise and begin printing.
```
  - [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1660778/Configuration.zip)

Here is my M503 as well:
```
Send: M503
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E100.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X400.00 Y400.00 Z8.00 E50.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000 Y2000 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P400.00 R1000.00 T1000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H190 B60 F0
Recv: echo:  M145 S1 H240 B90 F0
Recv: echo:PID settings:
Recv: echo:  M301 P21.00 I1.25 D86.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.80
Recv: ok
```

GMagician, 2018-01-24:
Not sure but tried with m420 s1 to enable bed leveling?

SirEggbert, 2018-01-24:
Ok I have added it into my start up code:

```
M280 P0 S160 ; BLTouch alarm release
M117 Auto Leveling ...
G4 P100 ; delay for BLTouch
M420 S1 ; Turn On Bed Leveling
G28 ; Home Axis
M502
G29 ; Z-Probe the bed
M500
M117 Priming Extruder ...
G0 X1 Y1 F3000 ; Go to front
G0 Z0.15 ; Drop to bed
G92 E0 ; zero the extruded length
G1 X100 E40 F500 ; Extrude 25mm of filament in a 4cm line
G92 E0 ; zero the extruded length
G1 E-5 F500 ; Retract a little
G1 X180 F500 ; Quickly wipe away from the filament line
M117 Printing ...
G1 Z5 ; Raise and begin printing.
```

Festivejelly, 2018-01-25:
Uhhh why is your bed showing -4mm from your home? Your mesh should be much closer to 0

Festivejelly, 2018-01-25:
DO you home in the centre of the bed?

thedragonlord71, 2018-02-10:
I have the same issue on my CR-10S. Any solution?

thinkyhead, 2018-02-17:
@thedragonlord71 — 
- After `M420 S1` do you see Z move when moving in X and/or Y?
- If that is working, then how far out is your leveling compensation?

If you see a discrepancy where one side is always low and the other side is always high, this most likely indicates a twisted X gantry.

thedragonlord71, 2018-02-17:
I've instead the M420 S1 in my gcode right  after the G29... When the printing is on I haven't noticed a Z change during X Y "Travels"....

thinkyhead, 2018-02-28:
@thedragonlord71 —

`M420 S1` is not needed after `G29` but it does no harm.

After `G29` what is the output of `M420 V`?

If you flash Marlin with `bugfix-1.1.x` with `DEBUG_LEVELING_FEATURE` enabled you can then use the `G29 Q` command to get extra information about points on the bed and how much leveling compensation is being applied. If you enable `M114_DETAIL` you can get further information about the current point using `M114 D`.

Try out these commands after your `G29`, with leveling enabled, and show us some of the output.

boelle, 2019-02-20:
@thedragonlord71 did you try what thinky mentioned?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek