MarlinFirmware/Marlin #9340: Manual levelling mesh not being applied to print - Hephestos i3

irondylis, 2018-01-24:
Hi
I'm having an issue with my Hephestos i3 (Marlin 1.1.8). I'm trying to use manual bed levelling without success. I've Googled everything I can think of and I'm stuck. Any help or pointers to settings I could try much appreciated.

So far I am able to do the manual bed levelling. The points get recorded (verified with M420 V) from Pronterface. However when I actually print the values are ignored. The Z axis doesn't make any compensation moves.

I've tried an extreme version by setting the levelling points to 0, 1mm, 2mm across a 9 point grid, expecting the z-axis to lift as it prints the first layer but it doesn't budge.

So far the settings I've tried changing are
#define PROBE_MANUALLY
#define MESH_BED_LEVELING
//#define MIN_SOFTWARE_ENDSTOPS
#define Z_MIN_POS 0

Please let me know what information you need to help me with this issue. I'm going out of my mind!!

Many Thanks
Stu

Its-my-right, 2018-01-25:
Try putting M420 S1 in your part g-code, after initial G-28 commands :)

irondylis, 2018-01-25:
You Sir/Madam are a bloody legend. :)
Thank you thank you thank you. This should be added to the documentation on manual bed levelling if it's not already there and I missed it. That was doing my nut for several days.
Cheers
Stu

closed 2018-01-25 by irondylis