MarlinFirmware/Marlin #9344: Marlin raises z to the moon for no reason

thehans, 2018-01-25:
### Bug Report
- Description: Running Marlin 1.1.8 on RAMPS 1.4.  After execution of gcode command containing movement only in XY axes, Marlin just raises Z axes until I reset the printer.
- Expected behaviour: don't raise z axis unless gcode says to
- Actual behaviour: Z axis raises
- Steps to reproduce:
I had trouble trying to print a particular gcode which came from slic3r prusa edition.  It starts the print by getting up to temperature, probing z, and putting down the start of a skirt line along x axis.  At the end of this line it goes off track and raises Z.  I was able to truncate the gcode down to just the first few commands where the issue occurs and I think its small enough to just display inline here.  I added some M0 wait command to help me single out the line that was causing the problem.
```
; generated by Slic3r Prusa Edition 1.38.5-prusa3d-linux64 on 2018-01-24 at 20:07:30

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm

M107
M190 S85
M109 S250
G28 		; home all axes
G1 Z5 F5000 ; lift nozzle
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.200 F48000.000
G1 E-2.00000 F2400.00000
G92 E0
G1 X21.213 Y79.126 F48000.000
G1 E2.00000 F2400.00000
G1 F1425.6
G1 X22.560 Y78.904 E2.04623
M0 Ready to crash Z?
G1 X197.442 Y78.904 E7.96576
M0 Message displays as Z crashes
G1 X198.087 Y78.972 E7.98770 ; never gets here
```
I should also mention that my travel speeds in slicer settings are much higher than "practical", but I limit axis speeds in firmware to reasonable values that I know the printer can handle.  I do this so that when I reduce feedrate(usually because I want to *extrude* slower), the travel moves are still fast enough that after scaling, they still move at max speed, since I think this helps with stringing etc.

I updated my firmware from a 2yr old RC to 1.1.8 a week or so ago, and it has made a dozen or so successful prints since the upgrade, until I ran into this strange problem.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1662476/Config.zip)

filipgoc, 2018-02-01:
This looks interesting. Sorry, no ideas so far, but I'm curious what bigger minds bring here.

thehans, 2018-02-01:
The part that messes it up has the printer moving a somewhat large X distance in a single line of gcode: 197.442 - 22.560 = 174.882mm travel

So I'm wondering if large travel (most of the bed for an average 3d printer) might have anything to do with the issue?

thehans, 2018-02-01:
Also I forgot to mention I am using UBL and was thinking it was also plausible that mesh data somehow got corrupted in such a way that it goes to infinity at some specific point?

ripp2003, 2018-02-01:
Seems like here, same problem : [https://github.com/MarlinFirmware/Marlin/issues/9405](url)

thehans, 2018-02-01:
Link is bad, you can just type #9405 as a shortcut by the way.

fr33dimension, 2018-02-01:
I have a similar issue, the mesh is offset. Once you hit the end of the mesh z is going up until you stop it.
In my case it has something to do with the probe offset. It doesn't account for that and thinks the bed stops where that probe can test. Didn't find a solution and went back to 1.1.4 which worked great and is again.

Squid116, 2018-02-01:
I have observed this behavior as well (z to infinity), but haven't yet narrowed down the conditions in which it occurs.

thehans, 2018-02-01:
Well the gcode I pasted is very repeatable for me, at least on my specific printer.  If someone has suggestions to troubleshoot / debug further, I'm willing to do some more testing to narrow it down.

Squid116, 2018-02-02:
@thehans - Not disagreeing with you, I was just adding some support to your ticket - i.e. you aren't the only one getting this.

Also worth noting: my mesh values are all < 0.1 mm away from 0 and I have seen this.

thehans, 2018-02-02:
I'm trying to add the report from running `G29 T0`

I sent `G29 T0` manually from RepetierHost but every time it either omits entire lines or at least some lines get cut off.  No idea what that's about or if the issue is related(could also be an error in the host software?), but I can't really get a single full report of the mesh points.
```
18:30:31.093 : Bed Topography Report:
18:30:31.117 : (0,7)                                                   (7,7)
18:30:31.138 : (15,191)                                                (191,191)
18:30:31.292 : -0.011    0.081    0.104    0.065    0.031    0.021
18:30:31.292 : 
18:30:31.416 : -0.075   -0.004    0.060    0.106    0.103    0.098    0.074    0.036
18:30:31.416 : 
18:30:31.540 : -0.029    0.043    0.085    0.095    0.111    0.094    0.095    0.085
18:30:31.540 : 
18:30:31.665 : -0.009    0.068    0.098    0.110    0.121    0.120    0.125    0.109
18:30:31.665 : 
18:30:31.791 : 0.041    0.106    0.136    0.148    0.150    0.148    0.156    0.145
18:30:31.791 : 
18:30:31.915 : 0.039    0.114    0.137    0.154    0.161    0.168    0.180    0.163
18:30:31.915 : 
18:30:32.041 : 0.030    0.096    0.135    0.164    0.176    0.184    0.196    0.180
18:30:32.041 : 
18:30:32.192 : [ 0.023]   0.075    0.121    0.164    0.177    0.187    0.197    0.174
18:30:32.214 : (15,15)                                                    (191,15)
18:30:32.238 : (0,0)                                                     (7,0)
18:30:48.625 : N17 G29T0*64
18:30:48.638 : echo:Home XYZ first
18:30:48.642 : Bed Topography Report:
18:30:48.664 : (0,7)                                                   (7,7)
18:30:48.688 : (15,191)                                                (191,191)
18:30:48.838 : -0.174   -0.090   -0.011    0.081    0.104    0.065    0.031    0.021
18:30:48.838 : 
18:30:48.963 : -0.075   -0.004    0.060    0.106    0.103    0.098    0.074    0.036
18:30:48.963 : 
18:30:49.087 : -0.029    0.043    0.085    0.095    0.111    0.094    0.095    0.085
18:30:49.087 : 
18:30:49.212 : -0.009    0.068    0.098    0.110    0.121    0.120    0.125    0.109
18:30:49.212 : 
18:30:49.337 : 0.041    0.106    0.136    0.148    0.150    0.148    0.156    0.145
18:30:49.338 : 
18:30:49.490 : 0.154    0.161    0.168    0.180    0.163
18:30:49.490 : 
18:30:49.615 : 0.030    0.096    0.135    0.164    0.176    0.184    0.196    0.180
18:30:49.615 : 
18:30:49.740 : [ 0.023]   0.075    0.121    0.164    0.177    0.187    0.197    0.174
18:30:49.762 : (15,15)                                                    (191,15)
18:30:49.786 : (0,0)                                                     (7,0)
18:30:54.626 : N18 G29T0*79
18:30:54.640 : echo:Home XYZ first
18:30:54.642 : Bed Topography Report:
18:30:54.666 : (0,7)                                                   (7,7)
18:30:54.688 : (15,191)                                                (191,191)
18:30:54.839 : -0.174   -0.090   -0.011    0.081    0.104    0.065    0.031    0.021
18:30:54.840 : 
18:30:54.964 : -0.075   -0.004    0.060    0.106    0.103    0.098    0.074    0.036
18:30:54.964 : 
18:30:55.089 : -0.029    0.043    0.085    0.095    0.111    0.094    0.095    0.085
18:30:55.090 : 
18:30:55.214 : -0.009    0.068    0.098    0.110    0.121    0.120    0.125    0.109
18:30:55.214 : 
18:30:55.339 : 0.041    0.106    0.136    0.148    0.150    0.148    0.156    0.145
18:30:55.339 : 
18:30:55.462 : 0.039    0.114    0.137    0.154    0.161    0.168    0.180    0.163
18:30:55.463 : 
18:30:55.616 : 0.096    0.135    0.164    0.176    0.184    0.196    0.180
18:30:55.616 : 
18:30:55.742 : [ 0.023]   0.075    0.121    0.164    0.177    0.187    0.197    0.174
18:30:55.763 : (15,15)                                                    (191,15)
18:30:55.787 : (0,0)                                                     (7,0)
```

thehans, 2018-02-02:
Here is the info from `G29 W`

```
18:48:59.121 : N19 G29 W*93
18:48:59.135 : echo:Home XYZ first
18:48:59.138 : Unified Bed Leveling System v1.01 active.
18:48:59.189 : Mesh 0 Loaded.
18:48:59.238 : UBL object count: 1
18:48:59.240 : planner.z_fade_height : 10.0000
18:48:59.243 : # of samples: 64
18:48:59.245 : Mean Mesh Height: 0.096406
18:48:59.245 : Standard Deviation: 0.072356
18:48:59.247 : zprobe_zoffset: -0.4000000
18:48:59.252 : MESH_MIN_X  ((((((220) / 2) - (220) / 2) + 15)>(0 + -29)?((((220) / 2) - (220) / 2) + 15):(0 + -29)))=15
18:48:59.256 : MESH_MIN_Y  ((((((220) / 2) - (220) / 2) + 15)>(-8 + -29)?((((220) / 2) - (220) / 2) + 15):(-8 + -29)))=15
18:48:59.284 : MESH_MAX_X  ((((((220) / 2) + (220) / 2) - (15))<(220 + -29)?((((220) / 2) + (220) / 2) - (15)):(220 + -29)))=191
18:48:59.291 : MESH_MAX_Y  ((((((220) / 2) + (220) / 2) - (15))<(220 + -29)?((((220) / 2) + (220) / 2) - (15)):(220 + -29)))=191
18:48:59.315 : GRID_MAX_POINTS_X  8
18:48:59.317 : GRID_MAX_POINTS_Y  8
18:48:59.343 : MESH_X_DIST  25.14
18:48:59.345 : MESH_Y_DIST  25.14
18:48:59.575 : X-Axis Mesh Points at: 15.000  40.143  65.286  90.429  115.571  140.714  165.857  191.000
18:48:59.784 : Y-Axis Mesh Points at: 15.000  40.143  65.286  90.429  115.571  140.714  165.857  191.000
18:48:59.785 : Kill pin on :41  state:1
18:48:59.836 : Unified Bed Leveling sanity checks passed.
```

fr33dimension, 2018-02-03:
I made the following observation. The mesh size is different in 1.1.8 compared to 1.1.4 even thought the config.h is the same. It is as if the probe offset is not taken into consideration, but if I change it to a higher number then 151 (bed is 194 offset x -43) it complains that the probe cannot get there. Here is the mesh info from 1.1.4 and 1.1.8 (below)
G29 I 999
M421 I 0 J 0 Z -0.084000 ; X 0.00, Y 0.00
M421 I 0 J 1 Z -0.280000 ; X 0.00, Y 66.67
M421 I 0 J 2 Z -0.280938 ; X 0.00, Y 133.33
M421 I 0 J 3 Z 0.000000 ; X 0.00, Y 200.00
M421 I 1 J 0 Z 0.007000 ; X 65.00, Y 0.00
M421 I 1 J 1 Z -0.280625 ; X 65.00, Y 66.67
M421 I 1 J 2 Z -0.281563 ; X 65.00, Y 133.33
M421 I 1 J 3 Z -0.000000 ; X 65.00, Y 200.00
M421 I 2 J 0 Z 0.039687 ; X 130.00, Y 0.00
M421 I 2 J 1 Z -0.238438 ; X 130.00, Y 66.67
M421 I 2 J 2 Z -0.267188 ; X 130.00, Y 133.33
M421 I 2 J 3 Z -0.000000 ; X 130.00, Y 200.00
M421 I 3 J 0 Z 0.240002 ; X 195.00, Y 0.00
M421 I 3 J 1 Z 0.040002 ; X 195.00, Y 66.67
M421 I 3 J 2 Z -0.160002 ; X 195.00, Y 133.33
M421 I 3 J 3 Z -0.1100002 ; X 195.00, Y 200.00

G29 I 999
M421 I 0 J 0 Z 0.943750 ; X 0.00, Y 0.00
M421 I 0 J 1 Z 0.870937 ; X 0.00, Y 65.00
M421 I 0 J 2 Z 1.055938 ; X 0.00, Y 130.00
M421 I 0 J 3 Z 1.139375 ; X 0.00, Y 195.00
M421 I 1 J 0 Z 1.194687 ; X 50.67, Y 0.00
M421 I 1 J 1 Z 1.089687 ; X 50.67, Y 65.00
M421 I 1 J 2 Z 1.241562 ; X 50.67, Y 130.00
M421 I 1 J 3 Z 1.300000 ; X 50.67, Y 195.00
M421 I 2 J 0 Z 1.386875 ; X 101.33, Y 0.00
M421 I 2 J 1 Z 1.260937 ; X 101.33, Y 65.00
M421 I 2 J 2 Z 1.346875 ; X 101.33, Y 130.00
M421 I 2 J 3 Z 1.402187 ; X 101.33, Y 195.00
M421 I 3 J 0 Z 1.548750 ; X 152.00, Y 0.00
M421 I 3 J 1 Z 1.416875 ; X 152.00, Y 65.00
M421 I 3 J 2 Z 1.564687 ; X 152.00, Y 130.00
M421 I 3 J 3 Z 1.625312 ; X 152.00, Y 195.00

thinkyhead, 2018-02-06:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch in case the issue is already fixed. If we already got it then you can close the issue. If you still see the bad behavior then we can investigate further.

kAdonis, 2018-02-07:
Ihad a similar problem and read about this behaviour several times #9147, #8723
I think the Nozzle is leaving the Probed Area (The Mesh) 
your Mesh Boundaries are Incorrectly computed, this Bug is since 1.1.6   (Issue #8723)
You could try (on a cartesian or coreXY) to uncomment the following lines in Configuration_adv.h

//#define MESH_MIN_X MESH_INSET
//#define MESH_MIN_Y MESH_INSET
//#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
//#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)

to

 #define MESH_MIN_X MESH_INSET
  #define MESH_MIN_Y MESH_INSET
  #define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  #define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)

as describrd in #8723 this helped me.
but UBL is still buggy, I dont use it any more

hamster65, 2018-02-11:
I have the same issue (marlin 1.1.8, ubl active). I seems when the nozzle goes outside of the probed area, the z axis just moves up. Since the model I was printing had a smaller base, that happened completely unexpected in mid print. In addition, the emergency stop button on my lcd did not react so I had to cut power.

filipgoc, 2018-02-11:
@hamster65 can you reproduce this? If you can, please test with the bugfix branch.

Roxy-3D, 2018-02-11:
> I have the same issue (marlin 1.1.8, ubl active). I seems when the nozzle goes outside of the probed area, the z axis just moves up. 

More specifically...   I think this is happening when the nozzle goes off the mesh.    I'm going to take a closer look at this.

hamster65, 2018-02-14:
@filipgoc: I just reproduced the problem by placing a model as far to the edge as possible in cura. The printer seems to rise the Z axis with maximum speed. This time however, I could do an emergency stop with the button on my lcd unit. It seems Marlin does not react to very short button presses.
I'll try to test with the bugfix branch next weekend.

thinkyhead, 2018-02-17:
Presumably this has been patched with c8d3069.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

thinkyhead, 2018-02-17:
Duplicate of #9147.

closed 2018-02-17 by thinkyhead