MarlinFirmware/Marlin #9352: Suggestion: Easy upgrade to STM32

povlhp, 2018-01-26:
I am a n00b on this project, but I am good at thinking out of the box.

If 32-bit is the way to go, and we are starting to use a HAL, shouldn't it be possible to use a cheap STM32F1 Arduino-nano like device to piggyback on the existing Arduino boards ?
Then my GT2560RevB could work as a relative dumb interface to the hardware (using serial or I2C comms).

Then we could all upgrade to 32-bit, new features etc for $3.

The software on the old GT2560 would then simulate some generic hardware, so that only the "driver code" would have to know about pinouts. 

Not sure if this is a realistic hack, if the boards even has available pins for serial comms, rather than USB->Serial

kevintong116, 2018-01-29:
Something like the klipper project?
https://github.com/KevinOConnor/klipper

thinkyhead, 2018-02-14:
When the hardware exists, we'd be happy to help integrate it into Marlin.

povlhp, 2018-02-15:
There are multiple solutions, depending on your goal.  I was thinking of Arduino-like STM32F103C8T6. CDurrent print $3.62 w free shipping.
https://www.banggood.com/STM32F103C8T6-Small-System-Board-Microcontroller-STM32-ARM-Core-Board-For-Arduino-p-1058299.html (not affiliate link).

Or ESP-01 and friends for same price or less - They have WiFi, 80MHz, but too much might be dedicated to WiFi. Considered using one as pure WiFi bridge, but now with linear advance the 115k is the limiting factor.

The cheap STM32 would give us 64k flash, 20k SRAM, 72MHz. So the planner and general gcode processing could happen here. Then we just need a way to communicate with the old board with all the IO. Klipper project reaches 100k+ steps/s. But not only that, we would also have much more processing power available. Since the upgrade is that cheap most people would be able to afford it.

thinkyhead, 2018-02-23:
That's a cute board! It's actually not that hard to just assign all the pins in some random/logical order and get Marlin to compile for any board with good library support. We have a few STM32's already in the works. But before we can add something to Marlin at minimum we need a pin map giving defined roles to all the pins.

povlhp, 2018-02-23:
What I was thinking was to use existing pin mapping from config.h /. Boards and have the 32-bit software talk to a generic set of pins. The HAL. Then have the ATMega stub Code receive and translate the values. So one 32-bit Target. And one receiver stub on the board that is different per board.

thinkyhead, 2018-02-24:
Voodoo! Please don't harm any chickens. I look forward to hearing of your results!

boelle, 2019-02-20:
@povlhp any news?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

povlhp, 2019-03-06:
I never got anywhere. Klipper does some of the same. Runs the printer board as dumb slave

boelle, 2019-03-06:
it's not to be rude that i do these rounds of cleanup

it's to get issues closed that are fixed, and also close those issues where people have left and not using marlin anymore. 

i think it's only fair that the dev people spend time on things where people can still test if what they are doing works

povlhp, 2019-03-06:
You did not understand me.
I agree to close it.
If I or someone else comes up with something that does this - Then we can
create a pull request and reopen. For the last few days I am on Klipper
again - But it is missing some commands, and has other issues. Klipper is
just an example, and maybe using klipper code on the controller would be an
idea.

boelle, 2019-03-06:
ahh, now i understand

but only you and @thinkyhead can close, i have not been given those powers :-D

closed 2019-03-06 by povlhp