MarlinFirmware/Marlin #9354: Feature Request : Retrieve Vals from Gcode

Skybound420, 2018-01-26:
I'm finding it may or may not be beneficial to my learning curve if my jerk and acceleration settings can be adjusted based on my print speed as determined in my slicer.  That said, is it possible to have the firmware scan the G code script(s) saved to the SD card for values to reference and use in math?  My current basic understanding is that Jerk is roughly 10% of print speed whereas Acceleration is roughly 10 times Jerk, so if Print Speed can be the baseline where other numbers are computed from, I would have a better bench mark to begin tweaking for my specific machine. 

Thank you.

GMagician, 2018-01-26:
I think this has something in common to #9350. I think it's a parser issue

Sebastianv650, 2018-01-27:
Jerk and acceleration are set to a value which satisfies your quality needs. Higher jerk or acceleration equals faster prints but may degrade print quality and vice versa. It depends on the stiffnes of your machine and your personal preferences. Not realy on print speed.

Skybound420, 2018-01-27:
Though I agree with your statement, I still feel these settings can be made variable depending on print speed and possibly other factors.  It's also very true that each machine is capable of different speeds based on rigidity, but then couldn't a multiplier be used that would govern how much the J/A settings are adjusted?  Perhaps it's just the way things work in my mind, but I'm comprehending these relationships to be very similar to the relationship between Esteps and Flow/Extrusion Multiplier.  The former being the crude setting that get's the process into range, the latter being the multiplier used to refine the former.

So maybe as an example it might read as; 

`X = 0;
Y = 0;
PRINT_SPEED = 0;
DEFAULT_ACCELERATION=PRINT_SPEED * X;
JERK = ACCELERATION/10 * Y;`

Sebastianv650, 2018-01-27:
It's not a question that it would be doable for SD card prints. But I realy don't get the meaning why someone would need that, in fact you are the first person I know requesting that.
Usualy, people want distinct acceleration on features which have distinct quality needs: low acceleration for perimeters, the highest possible for infill and support material. All this is usualy set by the slicer, as it's not detectable for Marlin.

A feature as you are requesting would be overwritten by the slicers settings or slicer settings would influence with this request..

Skybound420, 2018-01-27:
Perhaps its just my ignorance between how the firmware works from the G code file or vice-a-versa, but can you say that these settings already do vary in relation to print speed?  Is J or A used to determine the actual feed rate for each stepper?  

In order for me to better grasp the math, can you direct me to the .h and/or .cpp files that handle these determinations so that I might be able to better understand my own desire and possibly form better questions?

Sebastianv650, 2018-01-27:
> But can you say that these settings already do vary in relation to print speed? Is J or A used to determine the actual feed rate for each stepper?

At the moment they are fixed values, not varied by print speed. Jerk is used to limit the maximum force during arcs (=a lot of straight short lines), and this force isn't depending on the final speed of the segment. Therefore it makes no sense to set it according to max. print speed of a segment. You can think about jerk as the maximum speed change the printer is allowed to do without acceleration from stand still.
Acceleration is used to limit the force (force = mass * acceleration) during acceleration. Again, this force is not depending on final print speed.
Both forces has to be counteracted by the printers frame. As each force will try to flex the frame, you have to keep them in reasonable limits. There will be always a deflection, but if it gets too big you will see the deflection in the prints which is mostly not wanted.

> In order for me to better grasp the math, can you direct me to the .h and/or .cpp files that handle these determinations so that I might be able to better understand my own desire and possibly form better questions?

The planner calculates the speed profile (trapezoid) from junction speeds, nominal speed, jerk and acceleration. Then the blocks are stored in the block buffer. You can find it in planner.cpp and .h.
The stepper (stepper.cpp and. h) executes the calculated blocks.

Prepare to spend some days if you want to understand the planner in-detail.

Skybound420, 2018-01-27:
It will likely take me weeks to chew up, not days, but rather than waste your (and others) time on this issue, coupled with my own sheer desire to see the man behind the curtain, I will peruse in spare time and digest what I can.  I realize it's taken many far more experienced than I many years to get the firmware to where it is, but like anything and everything, I gotta start somewhere.

Thanks for spending your time explaining these things to me.

Roxy-3D, 2018-01-29:
@Skybound420    I would consider adding an MCode to let me set values.    I would have it set
```
  planner.max_jerk[X_AXIS] = DEFAULT_XJERK;
  planner.max_jerk[Y_AXIS] = DEFAULT_YJERK;
  planner.max_jerk[Z_AXIS] = DEFAULT_ZJERK;
  planner.max_jerk[E_AXIS] = DEFAULT_EJERK;
```
values to what I wanted to try.      (Those code lines up above are the only place the configuration.h specified values are referenced.  This is in the configuration_store.cpp file)   If you zap planer.max_jerk[] to a new value in your new MCode,  probably the new value will get picked up and used.

This is untested code...  But it is close to what I'm suggesting you add to Marlin_main.cpp:
```
  inline void gcode_M47() {
    if (parser.seen('X')) 
      planner.max_jerk[X_AXIS] = =  parser.value_float();
    if (parser.seen('Y')) 
      planner.max_jerk[Y_AXIS] = =  parser.value_float();
    if (parser.seen('Z')) 
      planner.max_jerk[Z_AXIS] = =  parser.value_float();
    if (parser.seen('E')) 
      planner.max_jerk[E_AXIS] = =  parser.value_float();
    }
```

closed 2018-02-14 by thinkyhead