MarlinFirmware/Marlin #9355: Micro Stutters

prahjister, 2018-01-26:
What is the best way to see what is going on at any time and debug my issue.
I am getting micro stutters/pauses.
It is related to the screen updates i think.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1668131/Marlin.zip)

I have adjusted delta segments lowered the refresh of the screen to the point its almost unusable. I tried on many different versions. All seem to have no affect.

If I print slower it appears to go away. But it may just be masking it. MKS Board.

Anyway to optimize Marlin? I have also tried adjusting platform.txt and modify gcc.

I am getting good prints do get me wrong. I know I should get a 32 bit board.

Sebastianv650, 2018-01-27:
Have a look at #9219, it might give you some details and even tipps for squeezing out some extra cycles.
Knowing we are already at the edge with prozessing power at cartesian machines, I'm impressed it's even possible to run deltas to be honest ;)

prahjister, 2018-01-28:
All my issues were when the xyz coordinates updated on the info screen. I am doing a test print now and after watching the first couple layers I think this fixes it. I reverted all my changes with segments per segment and update interval and went into ultalcd.cpp and changed 

```
  /**
   *
   * "Main" menu
   *
   */

  void lcd_main_menu() {
    START_MENU();
    MENU_BACK(MSG_WATCH);
 to 
```

```
  /**
   *
   * "Main" menu
   *
   */

  void lcd_main_menu() {
    START_MENU();
    defer_return_to_status = true;
    MENU_BACK(MSG_WATCH);

```
It seems to affect all menus and i don't have to go back to the info screen unless I have to. The only reason i like to go there is occasionally temperature and speed up or slow down prints

I wish i could turn off the xyz coordinate entirely

prahjister, 2018-01-28:
Finished print it is beautiful. Layed down plastic perfectly. Thank you for all your efforts. Marlin is King.

CCS86, 2018-02-03:
@prahjister 

You saw a big reduction in motion stuttering just from the added "defer_return_to_status = true;" line?

I am still fighting this on my cartesian machine. Printing concentric infill at 60 mm/s really falls apart:

https://www.youtube.com/watch?v=LOkfg-KbApI

prahjister, 2018-02-03:
https://www.cnx-software.com/2018/02/03/he3d-k280-delta-3d-printer-review-part-2-upgrades-calibration-and-more-3d-prints/

Here shows what I was dealing with. It may work with your issue as well

thinkyhead, 2018-02-06:
> Anyway to optimize Marlin?

We've made a huge amount of optimization to the LCD display update rate already, including only drawing the items visible in each segment, adding options to use hollow boxes instead of inverted text, and so on. If you're still experiencing issues then you may just be trying to do too many small segments, leaving the planner vulnerable to starvation.

What causes stuttering? Segments are being completed faster than they're coming in.

Your steps-per-mm settings are as modest as they come. Your delta should be able to keep up quite easily with well-formed G-code, assuming all is well otherwise.

You may enable `NO_VOLUMETRICS` and `NO_WORKSPACE_OFFSETS` for a tiny optimization effect.

Can you confirm you're not experiencing slow or problematic host communication, and that the SD card is in good order? You may find that SD performance increases with a low-level format of the SD card.

thinkyhead, 2018-02-17:
The character display should update pretty quickly, but the current code always updates the whole display. As we get to work on overhauling the display system following the 2.0 release, we plan to make the status screen only update values that have changed and seek out other possible optimizations.

There are two things which may lead to pauses or stutters. If for some reason the stepper interrupt is disabled for a long period of time, this could cause a pause. But we are very careful not to ever disable the stepper ISR for longer than a few µs. The other cause is a starved planner, which only occurs if there are not enough G-code commands arriving to keep up with the planner getting empty. In that case, a smaller `DELTA_SEGMENTS_PER_SECOND` can help.

You might also try setting retraction to 0 in your slicer profile and see if a print with no retraction exhibits this behavior. Then try different `BAUD_RATE` settings, different `BLOCK_BUFFER_SIZE`, and, various other random things. If you are able to find some change in settings that prevents the small stutter, this will help pinpoint whether stuttering is happening due to planner starvation or due to some other cause.

closed 2019-07-06 by shitcreek