MarlinFirmware/Marlin #9356: ### Compile Error BQ Hephestos 2

m50008, 2018-01-26:
------------------------------------------------------------
### Compile Error
When I compile with Hephestos 2 I get an error:
```
Arduino: 1.8.2 (Windows 10), Board: "Arduino Mega ADK"

In file included from sketch\Marlin_main.cpp:314:0:

sketch\endstop_interrupts.h: In function 'void setup_endstop_interrupts()':

endstop_interrupts.h:148: error: static assertion failed: Y_MIN_PIN is not interrupt-capable

       static_assert(digitalPinToPCICR(Y_MIN_PIN) != NULL, "Y_MIN_PIN is not interrupt-capable");

       ^

exit status 1
static assertion failed: Y_MIN_PIN is not interrupt-capable

```

GMagician, 2018-01-26:
Sure that pin is interrupt capable?
otherwise put // in front of ENDSTOP_INTERRUPTS_FEATURE in configuration.h

m50008, 2018-01-26:
It is the example config for my printer from the /Marlin/example_configurations/BQ/Hephestos_2/. I havent change anything there or should i?

GMagician, 2018-01-26:
Don't know about your board, sorry

Bob-the-Kuhn, 2018-01-26:
You've selected the wrong board.  Instead of "Arduino Mega ADK" select "Arduino/Genuino Mega or Mega 2560" and that error will go away.

m50008, 2018-01-27:
I tried both and non of them were working, I just did what Gmagican said and that works.

closed 2018-01-27 by m50008