MarlinFirmware/Marlin #9358: MKS Base 1.6 booting but not answering. How to get more details?

pgrandin, 2018-01-26:
### Bug Report
- Description: The board flashed with Marlin 1.1.8 boots but doesn't answer to any command
- Expected behaviour: Get replies to commands
- Actual behaviour: I get the startup banner, but nothing else after that.
- Steps to reproduce:
  - Flash board with v1.1.8

Here's the output from my minicom session:
```
start
echo:Marlin 1.1.8

echo: Last Updated: 2017-12-25 12:00 | Author: (none, default config)
echo:Compiled: Jan 22 2018
echo: Free Memory: 5518  PlannerBufferBytes: 1264
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00

M105
```
I'm replacing a RAMPS v1.4 with this MKS Base. With the RAMPS board, I was getting the expected behavior. None of the commands I've tried have produced an output. How can I troubleshoot further?

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1668650/Config.zip)

melvinisken, 2018-01-30:
Just in case: do you send a line feed at the end of your commands? For example, in the Arduino IDE this is disabled by default. I don't know about minicom.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

the1snm, 2018-03-14:
Was this confirmed to be an actual issue? I just flashed my MKS V1.4 last night with with 1.1.8 from 1.1.6 because nothing I could do would allow UBL to compile in 1.1.6 but that's another story. One of the first things I noticed with 1.1.8 is that it would not respond to Octoprint often when the printer would boot. I'd have to reboot Octoprint every time the printer rebooted and this never happened on 1.1.6. Sometimes it required rebooting both in order get it working. I was actually looking up some other items on UBL when I ran across this post so I figured I'd check in.

thinkyhead, 2018-03-14:
If there are bugs in 1.1.8 they are forever in 1.1.8. So, best to test the latest code that represents the future. https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip

boelle, 2019-02-20:
@pgrandin @the1snm 

Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek