MarlinFirmware/Marlin #9360: Bug: Last segment not joined

thinkyhead, 2018-01-26:
The planner has a bug where the last segment is not joined to the previous. It occurs on all machines, for all types of moves. The G-code below should produce 4 identical square moves with no stopping between corners. However, it always stops on the next-to-last segment before doing the last segment.

Chances are that recent changes to the planner around move chaining introduced this bug.

```gcode
G28
G1 Z5

M1 Click for Square

G1 X100 F4500
M400
G1 Y100
M400
G1 X0
M400
G1 Y0
M400

M1 Click for Square

G1 X50
G1 X100
M400
G1 Y50
G1 Y100
M400
G1 X50
G1 X0
M400
G1 Y50
G1 Y0
M400

M1 Click for Square

G1 X50
G1 X75
G1 X100
M400
G1 Y50
G1 Y75
G1 Y100
M400
G1 X50
G1 X25
G1 X0
M400
G1 Y50
G1 Y25
G1 Y0

M117 Bye!
```

Ping: @Bob-the-Kuhn @Sebastianv650 @tcm0116

Roxy-3D, 2018-01-27:
It is my belief the
```
  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVE
```
change is where this got broken.    It is also causing some weird pause issues on UBL enabled systems.   
In the case of a Cartesian machine running a mesh based bed leveling system, I do not believe it produces any benefit.    But it does add a significant amount of computation load to the AVR processors.

It is my request that we have a discussion of whether or not this should be enabled by default on Cartesian machines.   I'm having trouble getting to that conclusion.

tcm0116, 2018-01-27:
I don't think `SEGMENT_LEVELED_MOVE` is the root cause. I'm seeing pauses on my delta, which is using UBL and can't use `SEGMENT_LEVELED_MOVE`. I think there's definitely something going on in the planner. 

That said, I agree that `SEGMENT_LEVELED_MOVE` seems like an unnecessary feature. The mesh leveling systems already segment at the mesh boundaries, which is a that's needed for cartesian. For kinematic machines, it would be nice to sub-segment the moves between the mesh boundaries so that you don't end up with a tiny move at the boundary. UBL may do this, but I just haven't looked into it.

Roxy-3D, 2018-01-27:
> UBL may do this, but I just haven't looked into it.

Yes.  It does.  And in a computationally optimized way.

Sebastianv650, 2018-01-27:
I can't see the bug here. If you tell Marlin to wait until a move has finished (M400) this means you wait until the planner is drained. So the last move ends at speed 0. The next one will start with speed 0.
Or do I look at the wrong point and I don't got the meaning?

thinkyhead, 2018-02-05:
The G-code I posted above affects all moves, whether leveling is on or not.

@Sebastianv650 Did you run the G-code? The M400 are there to demonstrate that the segment just before the `M400` is not chaining to the previous segment at all.

It appears that a recent patch to the planner chaining is causing this effect.

thinkyhead, 2018-02-05:
> That said, I agree that `SEGMENT_LEVELED_MOVE` seems like an unnecessary feature.

It is necessary. Without doing additional segments between the two points, the planner will not do any interpolation, just a straight line, between those points. Lines will also tend not to be consistent. Yes, lines whose start and end points are between mesh points will get the bilinear interpolations at the ends of those lines, but not between them. The intent of using simpler small segmentation is to eschew the arbitrary division of lines on mesh boundaries and just let the interpolation take care of it. The smaller the segments are, the more closely the nozzle should follow the (bilinear) contours of the bed.

There's an argument to be made that bilinear interpolation is just a "best guess" and maybe the Z height would be better just doing straight lines between mesh boundaries. Also an argument can be made that both approaches are only approximate, and neither one is superior.

tcm0116, 2018-02-05:
>The smaller the segments are, the more closely the nozzle should follow the (bilinear) contours of the bed.

Except for the fact that the bilinear leveling correction is not a contour. It's a mesh that is linear between the points on the mesh.

>the Z height would be better just doing straight lines between mesh boundaries

Not better, exactly the same. A segment that runs from one corner of mesh box to the other will be the exact same if it is one segment or many segments with the only different being the processing necessary to stitch the multiple segments back into one.

Sebastianv650, 2018-02-05:
I can see now what you mean. In the following screenshot you can see the entry and exit steprates of each executed block from your second example (after second click). Each color change is one side of the square:

[image]

As the block is always empty at start, the first G1 is splitted into two parts. Therefore it starts at jerk speed (403) and accelerates to cruising speed. The second half of the G1 move is started at cruising speed and ended at MINIMAL_STEP_RATE (120). The second G1 move starts and ends with MINIMAL_STEP_RATE.
So the blocks are joined (exit = start of next block), but I know you expect to see cruising speed as final rate of the middle block and as start rate of last block (2nd G1).

I's not a new bug. I tried it with my good old Marlin 1.1 from 5.5.2017 with the same result. I think the problem will be around the forward or, more likely, the reverse planner function. But I found nothing obvious until now.

thinkyhead, 2018-02-05:
> Except for the fact that the bilinear leveling correction is not a contour.

Are you sure about that? Here's what bilinear interpolation looks like within a single grid square, where temperature represents magnitude:

[image]

thinkyhead, 2018-02-05:
> It's not a new bug. I tried it with my good old Marlin 1.1 from 5.5.2017…

That doesn't surprise me, but I am surprised it was never caught until now. So, yeah, I will be analyzing the hell out of that today!

tcm0116, 2018-02-05:
>Here's what bilinear interpolation looks like within a single grid square, where temperature represents magnitude

I stand corrected.

tcm0116, 2018-02-05:
Could it be because of the following in `Planner::reverse_pass`?

```
  if (movesplanned() > 3) {
    const uint8_t endnr = BLOCK_MOD(block_buffer_tail + 2); // tail is running. tail+1 shouldn't be altered because it's connected to the running block.
                                                            // tail+2 because the index is not yet advanced when checked
```

thinkyhead, 2018-02-05:
> Could it be because of the following in Planner::reverse_pass?

I will try variants of that and see! Also, I was going to consult Grbl, which has developed considerably since Marlin first forked from it.

Sebastianv650, 2018-02-05:
> That doesn't surprise me, but I am surprised it was never caught until now.

Well, usualy this will only happen at the end of the print, where it might be nearly always a z lift or moving the bed to the front. Not easy to notice it there.
I'm just adding it to the list with other long- lasting things like wrong speeds and accelerations in DELTAs (has to be quite obvious if someone uses a DELTA..) and a stepper which was able to execute blocks with no or outdated trapezoids. :)

tcm0116, 2018-02-05:
>has to be quite obvious if someone uses a DELTA

You'd think it would've been, but it's actually quite difficult to judge the speed of the nozzle. Probably the most noticable time was during homing, but everyone likely just used the default speeds and didn't pay much attention. I didn't actually notice it until I was digging though the linear advance code.

thinkyhead, 2018-02-05:
>  `if (movesplanned() > 3) {`

Changing this to 2 has no effect, but it doesn't seem to hurt either.

Preliminary success: The problem goes away if I comment out the two lines that set `BLOCK_BIT_START_FROM_FULL_HALT`, so possibly the conditions being used to set this flag are incorrect. Looking further.

thinkyhead, 2018-02-05:
By the way, those conditions are this…
```cpp
!moves_queued || UNEAR_ZERO(previous_nominal_speed)
```
…and if that fails, then after jerk is calculated, this:
```cpp
previous_safe_speed > vmax_junction_threshold && safe_speed > vmax_junction_threshold
```

tcm0116, 2018-02-05:
It sounds like we're back to the same problem that splitting the first
segment into two was trying to solve.

tcm0116, 2018-02-05:
>Changing this to 2 has no effect

Did you also change the next line to:

```
const uint8_t endnr = BLOCK_MOD(block_buffer_tail + 1);
```

thinkyhead, 2018-02-05:
Nevermind. My result was spurious. Something I did got rid of the behavior, but it wasn't preventing `BLOCK_BIT_START_FROM_FULL_HALT` from being set. I'll keep testing till I have something definitive.

thinkyhead, 2018-02-05:
Log of a single set of two moves in a straight line, with the first move split by the planner:
```
Set BLOCK_BIT_START_FROM_FULL_HALT due to moves_queued==0
recalculate()
reverse_pass()
  movesplanned==1
forward_pass()
  forward_pass_kernel()
  forward_pass_kernel()
recalculate_trapezoids()
calculate_trapezoid_for_block() entry==0.13 exit==0.00
recalculate()
reverse_pass()
  movesplanned==2
forward_pass()
  forward_pass_kernel()
  forward_pass_kernel()
  forward_pass_kernel()
recalculate_trapezoids()
calculate_trapezoid_for_block() entry==0.13 exit==1.00
calculate_trapezoid_for_block() entry==1.00 exit==0.00
recalculate()
reverse_pass()
  movesplanned==3
forward_pass()
  forward_pass_kernel()
  forward_pass_kernel()
  forward_pass_kernel()
  forward_pass_kernel()
recalculate_trapezoids()
calculate_trapezoid_for_block() entry==1.00 exit==0.00
calculate_trapezoid_for_block() entry==0.00 exit==0.00
```

thinkyhead, 2018-02-05:
Although the log simply shows a low step rate for the start and end of the last segment, it doesn't move more slowly. The last segment is just as fast. It simply acts like a separate accelerated move.

closed 2019-05-16 by thinkyhead